heitzmann/gdstk #95: gdstk.Label with write_oas

frankxxu, 2022-05-19:
Hi Heitzmann,
I have gdstk.Label rotation=numpy.pi/2, anchor='n' and magnification=0.1.  I used 'write_oas' to write to an oasis file, and read in Virtuoso.  None of these properties changed.  I ruled out virtuoso's issue by using 'write_gds' function, and it seems to be working fine.

I think that write_oas does not take label's properties correctly. 

Thanks,
-Frank

heitzmann, 2022-05-19:
@frankxxu Label anchor and transformations are not supported by the OASIS file format. I've updated the documentation in [f43736d](https://github.com/heitzmann/gdstk/commit/f43736d528441412c042398bb4868cb3effbebe0) to make this clear.

A possible solution would be to create a cell only to hold the label and rotate the cell.

frankxxu, 2022-05-19:
Is this performance issue or oasis structure does not support?

heitzmann, 2022-05-19:
The OASIS file format doesn't have support for label transformations and anchoring. That's only possible in GDSII (and, for visualization, in SVG).

frankxxu, 2022-05-19:
I see. Thanks.
-Frank

closed 2022-05-19 by frankxxu