heitzmann/gdstk #96: rename a cell as in gdspy

cecial, 2022-06-02:
Hi, with gdstk, how to rename a cellname? as in gdspy, we can do lib.rename(oldcell,newcell)
can as assign new name through cell.name attribute?

and how to replace a reference cell? as in gdspy, lib.replace_references(oldcell,newcell)

heitzmann, 2022-06-02:
These functions are not implemented in gdstk at this point. You can assign directly to `Cell.name` (`mycell.name = "new name"`), but if any of your references uses the cell name instead of an actual reference to the cell itself, it should also be reassigned (this is not usually the case, but it can happen). This search for possibly dangling references is what `rename` does in gdspy.

As for `replace_references`, a simple loop over the library should solve the issue. I can implement it when I have some fre time. Meanwhile, it should be more or less straightforward to copy the source from gdspy if you need it urgently.

cecial, 2022-06-03:
> These functions are not implemented in gdstk at this point. You can assign directly to `Cell.name` (`mycell.name = "new name"`), but if any of your references uses the cell name instead of an actual [...]

Thanks, heitzmann. Can you explain me what is 'if any of your references uses the cell name instead of an actual reference to the cell itself, it should also be reassigned'? a cell reference the cell itself?

after i rename the Cell.name, will all references to the Cell be updated automaticlly in the all library cells?
and if new cell name already exists in library, will there be a warning or an error, and rename is not successfully?

heitzmann, 2022-06-03:
When a [`Reference`](https://heitzmann.github.io/gdstk/library/gdstk.Reference.html#gdstk-reference) is created, it can use a `Cell`, a `RawCell`, or simply a cell name (the actual cell might not even exist, it's just a name). In the first case, renaming the cell will work fine, because the reference does not use the name, it holds a reference to the cell object itself. However, if the reference knows only the cell name, renaming the cell will not update the reference, so it will continue to point to the old name. For example:

```python
cell_a = gdstk.Cell("CELL A")
good_ref = gdstk.Reference(cell_a)
bad_ref = gdstk.Reference("CELL A")

cell_a.name = "renamed cell A"
```
After renaming, `good_ref` still holds a reference to `cell_a` (with its new name), but `bad_ref` will reference a `"CELL A"` by name, which does not exist anymore. You'd have to manually look for all references in the library that reference `"CELL A"` by name and rename them or (better yet) point them to `cell_a`.

cecial, 2022-06-03:
THanks， i got it~

heitzmann, 2022-06-07:
@cecial I've added a `Library.rename_cell` function to the Python API. Creating a `Library.replace_references` is less straightforward, but you can probably use a combination of renaming and [`Library.replace`](https://heitzmann.github.io/gdstk/library/gdstk.Library.html#gdstk.Library.replace) with similar results.

closed 2022-06-07 by heitzmann

cecial, 2022-06-08:
@heitzmann , in a case that CellA is referenced by CellTop, now i want to replace this reference with CellB, how can i do that with Library.replace?

cecial, 2022-06-09:
> Here's one way to do it:
> 
> ```python
> import gdstk
> 
> cell_a = gdstk.Cell("A")
> cell_a.add(gdstk.Label("A", (0, 0)))
> 
> cell_top = gdstk.Cell("TOP")
> cell_top.add(gdstk.Reference(cell_a))
> 
> lib = gdstk.Library("LIB")
> lib.add(cell_a, cell_top)
> 
> print(lib.top_level()[0].get_labels()[0].text)
> 
> cell_b = gdstk.Cell("B")
> cell_b.add(gdstk.Label("B", (0, 0)))
> 
> lib.rename_cell("A", "B")
> lib.replace(cell_b)
> 
> print(lib.top_level()[0].get_labels()[0].text)
> ```

Hi, heitzmann
here cell_b is created outside the library, if I'm dealing with a exist gds file, A and B already exists, what happens when rename A to B as B already exist?
in this case, how to replace A with B?

cecial, 2022-06-10:
Thanks， got it~~~