heitzmann/gdstk #98: cells Attribute in gdstk.Library

cecial, 2022-06-07:
Hi, in gdspy, lib.cells is a dict, whose key is cellname and value is gdspy.Cell object, this make i can easily access the cell object by lib.cells[cell].

in gdstk, lib.cells is a list containing all the gdstk.Cell objects, i need to use a loop to get the a particular cell object by its name, can I get it using a easier way?

heitzmann, 2022-06-07:
If you need to find cells over and over, you can build a dictionary for faster access:

```python
cell_dict = {c.name: c for c in lib.cells}
```

We don't use dictionaries by default for efficiency and consistency, since renaming a cell would mean changing the dictionary as well.

cecial, 2022-06-08:
Thanks for this update, this way is good~

closed 2022-06-08 by cecial