heitzmann/gdstk #101: uninitialized last_ctrl was used in Curve::cubic_smooth() and Curve::quadratic_smooth()

wylecho, 2022-07-18:
My code is as below:
```
gdstk::FlexPath* p_path = (gdstk::FlexPath*)gdstk::allocate_clear(sizeof(gdstk::FlexPath));
memset(p_path, 0, sizeof(gdstk::FlexPath));
p_path->simple_path = true;
p_path->init(head_point, 1, path_width, 0, curve_tolerance, tag); // head_point is of type gdstk::Vec2
p_path->cubic_smooth(tail_points, NULL, NULL, false); // tail_point is of type gdstk::Array<gdstk::Vec2>
```
When I debug into the underline code, I found the variable last_ctrl is (0, 0) instead of a meaningful value in the first iteration of for statement:
```
void Curve::cubic_smooth(const Array<Vec2> points, bool relative) {
    Vec2 last_point = point_array[point_array.count - 1];
    const Vec2* point = points.items;
    if (relative) {
        const Vec2 ref = point_array[point_array.count - 1];
        for (uint64_t i = 0; i < points.count - 1; i += 2) {
            Vec2 first_point = last_point;
            Vec2 smooth_ctrl = last_point * 2 - last_ctrl;
            last_ctrl = ref + *point++;
            last_point = ref + *point++;
            append_cubic(first_point, smooth_ctrl, last_ctrl, last_point);
        }
    } else {
        for (uint64_t i = 0; i < points.count - 1; i += 2) {
            Vec2 first_point = last_point;
            Vec2 smooth_ctrl = last_point * 2 - last_ctrl;
            last_ctrl = *point++;
            last_point = *point++;
            append_cubic(first_point, smooth_ctrl, last_ctrl, last_point);
        }
    }
}
```

heitzmann, 2022-07-18:
That's expected. What would be the “meaningful value” for the last control point when no control point has been defined (in the first iteration)? That information van only come from the previous curve segment or be manually input.

wylecho, 2022-07-19:
Understand. Thanks for your quick reply!

closed 2022-07-19 by wylecho