heitzmann/gdstk #102: Gds

cecial, 2022-07-20:


heitzmann, 2022-07-20:
@cecial GdsWriter is supposed to be a very low-level API for incrementally creating particularly large files.
Doing a blind recursion could insert a cell more than once in the final file, so some bookkeeping is required, which is out of the scope of this API.

For your example, recursion should be easy enough to implement at user level with [`Cell.dependencies`](https://heitzmann.github.io/gdstk/library/gdstk.Cell.html#gdstk.Cell.dependencies):
```python
lib = gdstk.read_gds(some_gds_file)
writer = gdstk.GdsWriter("new.gds")
writer.write(lib.cells[0], *lib.cells[0].dependencies(True))
writer.close()
```

cecial, 2022-07-20:
> 

Hi, Heitzmann
this implementation also meets my request, thanks!

closed 2022-07-20 by heitzmann