heitzmann/gdstk #103: gdspy to gdstk upgrade utility

tvt173, 2022-08-16:
Hi @heitzmann , I've been really happy working with gdstk lately. In some places I see some really major performance gains. It would be great to start moving more scripts and utilities from gdspy to gdstk. Do you happen to have a list of API differences between the two? It seems like the gap has closed quite a bit since the onset of the project, and most of the API is now fairly similar.

And it would be really awesome to have a utility that could upgrade a project built on gdspy to gdstk. I'm wondering if you've considered the possibility. I'm sure you must have a lot of scripts also you've converted or have considered converting 🙂

heitzmann, 2022-08-20:
Hi @tvt173 that's great to hear! Unfortunately I don't have such an utility to automate the process… I've considered it, but I'm not sure I'd be able to make it work on all cases, which might lead to a worse situation.

The list of API differences is indeed short, but I don't have it on the top of my head, sorry. It also changes a little from version to version…

tvt173, 2022-08-22:
hi @heitzmann , ok thanks. even if we don't have an automatic upgrade utility, it may be helpful to have a section in the docs noting API differences between the latest gdspy and gdstk for those who wish to upgrade. not the highest thing on my list of priorities right now, but if I get to trying to convert anything, I'll do my best to report what I find back to this issue. would be helpful if others do the same!

tvt173, 2022-08-23:
i wonder if we could use the VSCode python language server to do this reliably
https://code.visualstudio.com/api/language-extensions/programmatic-language-features#find-all-references-to-a-symbol

tvt173, 2022-08-23:
https://code.visualstudio.com/api/language-extensions/programmatic-language-features#rename-symbols

heitzmann, 2023-04-01:
It's been a while since this issue was opened and I really can't see myself having the time to dedicate to it, so I'm closing it for now. Feel free to open a PR if anyone wants to give it a shot.

closed 2023-04-01 by heitzmann

tvt173, 2023-04-03:
also check out this [PR](https://github.com/gdsfactory/gdsfactory/pull/833) in gdsfactory for an example of how a large library was ported from gdspy to gdstk. i wouldn't be surprised if GPT-4 does a reasonable job of this too now 😄