heitzmann/gdstk #105: Bug in fracture-while-rendering step

95-martin-orion, 2022-08-29:
The following code misbehaves in the rendering step:

<details>

```python
import gdstk

# has ~15k points, will require fracturing
circle = gdstk.ellipse((0, 0), 50, tolerance=1e-6)
triangle = gdstk.Polygon([[0,0],[0,80],[80,0]])
ct = gdstk.boolean(circle, triangle, "or")[0]
rect = gdstk.rectangle((200,40), (240, 100))
fill = gdstk.rectangle((-100, -100), (300, 150), layer=1, datatype=1)
holes = gdstk.boolean(fill, [ct, rect], "not", layer=2, datatype=2)

lib = gdstk.Library()
cell = lib.new_cell("Main")
cell.add(ct, rect, *holes)
lib.write_gds(f"{output_dir}/test.gds", max_points=2048)
```

</details>

The expected output is a rectangle with one rectangular hole and one circle+triangle hole, fractured as necessary:

<details>

[image]

</details>

In the actual output, the fracture edge connecting the circle+triangle hole to the rectangle stops short of the triangle edge:

<details>

[image]

</details>

Notably, the polygons in the pre-rendering cell are correct; it is only during rendering that the issue appears. Testing suggests that this might be a 32-bit integer overflow issue present in the "fracture-while-rendering" step but not in the normal `Polygon.fracture()` method, but I wasn't able to confirm this with certainty.

Fracturing polygons manually is a reasonable workaround until this is resolved. For example, adding this line right after defining `holes` in the script above will produce the correct output:
```python
holes = [x for p in holes for x in p.fracture(2048)]
```

heitzmann, 2022-09-04:
@95-martin-orion Thanks for pointing this out! I've included a fix that should work in your case and not significantly change existing layouts.

95-martin-orion, 2022-09-21:
Can you verify that this fixes the sample code above? On a related note, do you know why this was only affected the rendering fracture step, and not manual fracturing? IIUC, the code in the PR is used in both flows.

FWIW, my understanding is that a similar issue affects gdspy ([relevant code](https://github.com/heitzmann/gdspy/blob/21e5ad4391b6834d2ef8ece71214f28e2b8f88be/gdspy/clipper.cpp#L4780)).

heitzmann, 2022-11-02:
Yes, the example seems to be working. The difference you saw in both steps is due to default values of precision in different functions.

closed 2022-11-02 by heitzmann