heitzmann/gdstk #113: bounding_box() return a very small float number

cecial, 2022-09-22:
hi, I am using cell.bounding_box function, but find it returns a very small float number, like following
((0.0,-4.440892098500626e-16), (174.142, 68.26))

which is supposed to be ((0.0,0.0), (174.142, 68.26))

is it a python issue? or gdstk issue? or can we round it to the gds library resultion?

heitzmann, 2022-09-24:
The internal library uses doubles to store coordinate information, so that is the precision for all functions. This is expected to happen. Once you save a GDSII or OASIS file, rounding will automatically happen.

nmz787-intel, 2022-11-01:
Any update on how to overcome this internal non-rounding?

I am hitting an issue where I load a GDS, the values are have extreme number of decimal places, thus testing the polygons with `any_inside` fails because the code fails to see things as being line-on-line.

i.e. this is what gdstk loads my GDS as:
```
[[4.61852778e-14 1.00000000e+00]
 [9.08000000e+00 1.00000000e+00]
 [9.08000000e+00 3.53600000e+00]
 [4.61852778e-14 3.53600000e+00]]
```

but after saving it and then reloading the gdstk-authored GDS file, I get these coordinates:
```
[[0.    1.   ]
 [9.08  1.   ]
 [9.08  3.536]
 [0.    3.536]]
```

which works as expected when calling `any_inside`.

```
import gdstk

frame = [gdstk.Polygon([[0.   ,  4.636],
                        [-0.1  ,  4.636],
                        [-0.1  , -0.1  ],
                        [ 0.   , -0.1  ]]) ]

# works
other = gdstk.Polygon([
                       [0. ,   1.   ],
                       [9.08,  1.   ],
                       [9.08,  3.536],
                       [0.,    3.536]])

# fails
other = gdstk.Polygon([
                     [4.61852778e-14, 1.00000000e+00],
                     [9.08000000e+00, 1.00000000e+00],
                     [9.08000000e+00, 3.53600000e+00],
                     [4.61852778e-14, 3.53600000e+00]
    ])

lib = gdstk.Library()
cell = lib.new_cell("TEST")
[cell.add(f) for f in frame]
cell.add(other)
lib.write_gds(f'ughh.gds')

def gdstk_interacting(rects1, rects_to_return_status_of):
    results = []
    for rect in rects_to_return_status_of:
        if gdstk.any_inside(rect.points, rects1):
            results.append(rect)
    return results

touching_bounds = gdstk_interacting(frame, [other])
assert touching_bounds

```
[image]

nmz787-intel, 2022-11-01:
for now it seems I can work around the internal rounding (imprecision?) by this sequence:
```
lib = gdstk.read_oas(input_file_path)
tempfile = 'temp_for_rounding_error_workaround.oas'
lib.write_oas(tempfile)
lib = gdstk.read_oas(tempfile)
```
I first tried all sorts of values for the `read_oas` function's `tolerance` arg... but nothing seemed to work ( 1e-0, 1e-1, 1e-13, 1e-14).

heitzmann, 2022-11-01:
> Any update on how to overcome this internal non-rounding?

Sure, you can manually round the polygons yourself (if I'm not mistaken, numpy might have functions for that already as well).

closed 2022-11-01 by heitzmann