heitzmann/gdstk #114: HOWTO/feature-ask: boolean operations returning whole (original) polygons only

nmz787-intel, 2022-09-29:
for example, in the Boost polygon library, the resulting polygons are whole polygons originating in the first operand (or maybe it is the second, I get it mixed up). Is there an existing way to prune incomplete polygons when performing, ex: AND between a group and a selection box? (i.e. click-and-drag to select pre-existing shapes in powerpoint, google slides, etc...)

heitzmann, 2022-10-01:
I don't think I understand your use case. Could you explain it in more detail? Maybe include an image to show what you mean?

nmz787-intel, 2022-10-27:
[image]

heitzmann, 2022-10-28:
I see what you want, but that operation isn't exactly what polygon clipping is for, so implementing something like that would have to be a different operation. It can be easily accomplished with [`gdstk.any_inside`](https://heitzmann.github.io/gdstk/geometry/gdstk.any_inside.html#gdstk-any-inside), though: simply check whether each polygon in the first set has any points inside the "selection" polygon. Note that, if the selection polygon is always an axis-aligned rectangle, you can definitely implement a more optimized version of the test by hand.

closed 2022-11-02 by heitzmann