heitzmann/gdstk #116: Image export at specific location

JamesWhitehead5, 2022-10-04:
Enhancement/Feature request

Klayout has methods zoom_box and save_image which allow for exporting images of a design file at specific locations.

It seems like the write_svg method can only export whole cells, and cannot pick the center and extent of export. It isn't feasible to crop the SVG after acquisition due to file size.

Could we get a method take takes a view box argument, and exports a raster image at that location?

heitzmann, 2022-10-28:
You can create the rectangle that represents your view box in a temporary cell, reference the cell(s) you want to output into the same temporary cell, perform a boolean operation to clip all polygons with the view box, and export the resulting temporary cell.

heitzmann, 2022-12-11:
I'll leave the following minimal implementation as an example. Note that with this implementation, the viewport doesn't have to be a rectangle, any shape(s) will do.
```python
def cropped_svg(cell, viewport, outfile, *svg_args, precision=1e-3):
    temp_cell = gdstk.Cell("cropped")
    for poly in cell.get_polygons():
        if gdstk.any_inside(poly.points, viewport):
            temp_cell.add(
                *gdstk.boolean(
                    poly, viewport, "and", precision, poly.layer, poly.datatype
                )
            )
    temp_cell.write_svg(outfile, *svg_args)
```

closed 2022-12-11 by heitzmann