heitzmann/gdstk #119: Cell inheritance causes segmentation faults

joamatab, 2022-10-12:
How can we get rid of segmentation fault?

```
import gdstk

class Component(gdstk.Cell):
    pass

def rectangle():
    r1 = Component("FIRST")
    rect = gdstk.rectangle((0, 0), (2, 1))
    r1.add(rect)

    r2 = gdstk.Cell("r2")
    rect = gdstk.rectangle((0, 0), (2, 2))
    r2.add(rect)

    c = gdstk.Cell("Combined")
    ref1 = gdstk.Reference(r1)
    ref2 = gdstk.Reference(r2)
    c.add(ref1)
    c.add(ref2)

    c.flatten()
    return c

if __name__ == "__main__":
    import gdsfactory as gf

    lib = gdstk.Library()

    c = rectangle()
    lib.add(c)

    for ref in c.references:
        lib.add(ref.cell)
    lib.write_gds("a.gds")

    gf.show("a.gds")
```

https://github.com/amccaugh/phidl/pull/165

#117

@tvt173
@basnijholt
@ahoenselaar

tvt173, 2022-10-12:
to be clear, we're pretty sure this is the root cause of segmentation faults on https://github.com/amccaugh/phidl/pull/165 also, as `Device` inherits from `Cell` in a similar manner

joamatab, 2022-10-18:
Cc'ing 

@thomasdorch
@skandanc
@proppy

joamatab, 2022-10-26:
This was fixed by Andreas' reference counting fix
@ahoenselaar

we are still getting some segmentation faults on the gdsfactory migration from gdspy to gdstk. See branch below

https://github.com/gdsfactory/gdsfactory/tree/gdstk

closed 2022-10-26 by joamatab