heitzmann/gdstk #121: Use a binding generator for the Python interface

ianwilliamson, 2022-10-27:
The current Python interface is defined manually, which is a fairly error-prone approach. #117 is an example of the type of issue that can arise in manually defined Python wrappers, and there may be other similar bugs lurking in the current implementation. It would be worth considering the use of a Python binding generator, such as CLIF [[1](https://github.com/google/clif), [2](https://google.github.io/clif/clif/python/primer.html)].

pwuertz, 2022-10-28:
Tossing [pybind11](https://github.com/pybind/pybind11) into the ring ;)

heitzmann, 2022-10-28:
Sorry, but I have no interest or time to dive into this issue. If anyone wants to try and get it done, I could accept a PR to replace the existing bindings as long as the person accepts to keep maintaining the work.

closed 2022-12-18 by heitzmann