heitzmann/gdstk #122: segmentation faults

joamatab, 2022-10-27:
How can we remove the segmentation faults from gdstk?

when porting phidl from gdspy to gdstk and running `pytest` we get segmentation faults
https://github.com/joamatab/phidl/tree/gdstk

we thought it may be due to the `gdstk.Cell` inheritance. So we tried porting gdsfactory to gdstk using `gdstk.Cell` as a property instead of using inheritance. However `pytest` still causes segmentation faults

https://github.com/gdsfactory/gdsfactory/tree/gdstk

If you disable the python garbage collector you can get rid segmentation faults

@tvt153
@skandanC
@ahoenselaar

pwuertz, 2022-10-28:
For reference, the opt-in solution in pybind11 is creating `shared_ptr` bindings for unified reference counting - in case both the library internally and Python want to manage references to library objects.

https://pybind11.readthedocs.io/en/stable/advanced/smart_ptrs.html#std-shared-ptr

heitzmann, 2022-10-28:
@joamatab I can certainly look into the reference counting issue, but starting from the test suite in phidl or gdsfactory is not really helpful. Is there any chance you have a more concise example where you see a segfault? Any self-contained script that I can use as a starting point to figure out where the issue might be would be a better start.

joamatab, 2022-11-06:
Hi Lucas,

See example below to reproduce a segmentation fault, you may need to run more than once to see it

```
import numpy as np
import gdstk
from gdstk import Polygon

class MyPolygon(gdstk.Polygon):
    """Polygonal geometric object.

    Args:
        points : array-like[N][2]
            Coordinates of the vertices of the Polygon.
        gds_layer : int
            GDSII layer of the Polygon.
        gds_datatype : int
            GDSII datatype of the Polygon.
        parent : cell that polygon belongs to.

    """

    def __init__(self, points, gds_layer, gds_datatype, parent):
        """Initialize polygon."""
        self.parent = parent
        super().__init__(points=points, layer=gds_layer, datatype=gds_datatype)

if __name__ == "__main__":
    p1 = MyPolygon(((0, 0), (0, 1), (1, 0)), 0, 0)
    p2 = MyPolygon(((2, 0), (2, 1), (1, 0)), 1, 1)

    c = gdstk.Cell("tree1")
    c.add(p1)
    c.add(p2)

    c.flatten()
    c.remove(p2)
    print(len(c.polygons))

```

heitzmann, 2022-12-11:
@joamatab The issue here happens when the construction of `p1` raises an error due to the missing initializer for `parent`. There's no function in the `gdstk` source that is called at this point, so the issue might be with the definition of the `gdstk.Polygon` type and how it deals with inheritance. I'm investigating the issue when I can, so this is just to keep an update log.

lucas-flexcompute, 2022-12-17:
@joamatab I found 2 issues that might be the cause of these segfaults during deallocation. The first is a misuse of the object deallocator directly in subtypable types (0d8f730). The second was cleaning up partially-initialized objects in case of failed initialization (b146c8e). I'll prepare a minor release with the fixes later, but if you could take a look to see if they improve your experience, let me know!

heitzmann, 2022-12-17:
[@joamatab](https://github.com/joamatab) I found 2 issues that might be the cause of these segfaults during deallocation. The first is a misuse of the object deallocator directly in subtypable types ([0d8f730](https://github.com/heitzmann/gdstk/commit/0d8f7307fae54c6a18d3d3c07d012fc57510ceee)). The second was cleaning up partially-initialized objects in case of failed initialization ([b146c8e](https://github.com/heitzmann/gdstk/commit/b146c8e515a8fe5dc9bd3d22dc00242882c067d1)). I'll prepare a minor release with the fixes later, but if you could take a look to see if they improve your experience, let me know!

closed 2023-06-25 by heitzmann