heitzmann/gdstk #124: pip install gdstk on Windows?

joamatab, 2022-11-10:
Hi Lucas,

How can we `pip install gdstk` on Windows?

[image]

I think you need to trigger a release for the PyPi 

@skandanC
@pwuertz

joamatab, 2022-11-11:
Same issue happens on CentOs

[image]

joamatab, 2022-11-11:
#125  includes a fix for this

SkandanC, 2022-11-11:
Running the command that's on `pypi-packages.yml` for windows: 
[image]

chucks out this error:
[image]

Modifying the compiler version number puts out this error:
[image]

joamatab, 2022-11-11:
I was hoping the windows image on github actions has zlib installed

@basnijholt

pwuertz, 2022-11-11:
@joamatab

> How can we pip install gdstk on Windows?

The CI script for producing Windows wheels is ready, but there hasn't been a release (new version tag) to trigger it yet.
So as soon as a new version is tagged we should be able to pip install gdstk.

> Same issue happens on CentOs

As the project readme says, on Linux you can install gdstk as a conda package or build it from source. Building from source is easy on Linux, just install the listed requirements using your preferred package manager. The error message you posted suggests that "cmake" is missing on your system.

pwuertz, 2022-11-11:
@SkandanC

A few lines prior to the conan install command you found in `pypi-packages.yml`, note that MSVC is being selected as the default compiler, which is the compiler the version-option I added was intended for.

On your system conan can't find MSVC (not installed or environment not set up) and found GCC/mingw as a fallback. Usually, this is not the toolchain you'd use to build DLLs with, since the official Python binaries are built with MSVC.

I don't know if ABI compatibility is still an issue nowadays, but if GCC is the toolchain you want you can still give it a try. The next error you encountered just states that there are no pre-built conan packages for your toolchain. You can tell conan to build zlib from source by adding the `--build=missing` option on install.

joamatab, 2022-11-12:
Im trying to get the wheels build using CICD
https://github.com/pypa/cibuildwheel

joamatab, 2022-11-14:
the wheels work now for windows and MacOs

[image]

https://pypi.org/project/gdstk/#files

CentOs still not working yet

joamatab, 2022-11-14:
For CentOs to work the main issue we were getting is the zlib being older than 1.2.9

tvt173, 2022-11-14:
> For CentOs to work the main issue we were getting is the zlib being older than 1.2.9

so, i would expect it should still work on CentOS with conda, right? but not pip? still a very nice improvement to have it working for windows and mac!

joamatab, 2022-11-14:
yes, with conda it works on every OS

pwuertz, 2022-11-19:
@joamatab You might be able to use the same strategy I used for building gdstk on Windows CI:

- (pip) install conan
- Let conan fetch or build a static zlib of given version + create a cmake toolchain file
- Let pip install gdstk from source, with the cmake toolchain file specified via ENV

-> The gdstk cmake build process should be able to see the new zlib version

heitzmann, 2022-12-11:
With the recent release automation introduced by @joamatab, the package's available through PyPI. I'll consider this issue closed for now.

closed 2022-12-11 by heitzmann