heitzmann/gdstk #132: How can we get gdstk.points

joamatab, 2022-11-12:
```
import gdstk

p = gdstk.FlexPath(width=0.5, points=[(-0.1, 0), (+0.1, 0)])
print(p.points)
```

returns
```
AttributeError: 'gdstk.FlexPath' object has no attribute 'points'
```

ottumm, 2022-11-12:
@joamatab I'm not sure if this is what you're asking, but if you want to get the polygonal representation of the path, you can do that with `p.to_polygons()`.

The input parameters `points` and `width` [don't appear to be exposed as attributes](https://heitzmann.github.io/gdstk/geometry/gdstk.FlexPath.html#gdstk-flexpath).

joamatab, 2022-11-13:
Thank you Mike, yes, I'm using that trick too

How could expose points and width directly from Polygon?

ottumm, 2022-11-14:
Depending on your use case, you may be able to use `FlexPath.spin()`, `FlexPath.path_spines()`, and `FlexPath.widths()` to do what you want. Note that `FlexPath` objects can be quite complicated because each object can consist of multiple paths.

heitzmann, 2022-11-16:
@joamatab: What @ottumm said is probably your best option indeed!

closed 2022-12-11 by heitzmann