heitzmann/gdstk #133: What's happening with all these `0.9.*` releases?

ottumm, 2022-11-12:
Hi — I'm curious about what's going on with the many releases that have happened over the past two days?

@heitzmann, it looks like @joamatab is a new maintainer of this repo, is that correct?

I'm asking out of curiosity and also as someone working on a team that relies on this repo and saw our automated dependency updates blow up. Thanks!

joamatab, 2022-11-13:
Hi, yes, I've been trying to release the wheels automatically for Linux, MacOS and windows so it's easier to pip install gdstk

We started to use gdstk as the gdsfactory backend so I offered to help with the maintenance of gdstk

I hope I can finish it tomorrow, help with this is also most welcomed :)

ottumm, 2022-11-14:
Gotcha, thanks @joamatab, appreciate the context!

closed 2022-11-14 by ottumm