heitzmann/gdstk #137: remove layers from cell and references

joamatab, 2022-12-03:
How can we remove layers from a cell and all its references?

This code is supposed to remove layer (1,0)

```
import gdstk

cell = gdstk.Cell("CELL_parent")

cell_child = gdstk.Cell("CELL_child")
poly1 = gdstk.Polygon([(0, 0), (1, 0), (0.5, 1)], layer=2)
cell_child.add(poly1)
poly2 = gdstk.Polygon([(1, 0), (1.5, 1), (0.5, 1)], layer=1)
cell_child.add(poly2)

ref = gdstk.Reference(cell_child, columns=1, rows=1, spacing=(8, 10))
cell.add(ref)
cell.filter([(1, 0)], remove=True)
# cell_child.filter([(1, 0)], remove=True) # this works

gdspath = "a.gds"
lib = gdstk.Library()
lib.add(cell)
lib.add(*cell.dependencies(True))
lib.write_gds(gdspath)

import gdsfactory as gf
gf.show(gdspath)
```

[image]

heitzmann, 2022-12-08:
Hi @joamatab, this should work:

```python
# Insert original cell in the list so that we don't have 2 places to change arguments
for c in [cell] + cell.dependencies(True):
    c.filter([(1, 0)], remove=True)
```
or, if you prefer
```python
filter_args = ([(1, 0)], True)
cell.filter(*filter_args)
for c in cell.dependencies(True):
    c.filter(*filter_args)
```

closed 2022-12-11 by heitzmann

joamatab, 2022-12-11:
yes, that worked great, thank you Lucas

https://github.com/gdsfactory/gdsfactory/pull/982