heitzmann/gdstk #140: integer based representation

joamatab, 2023-01-04:
How could we move from floating point to integer based?

GDS database is integer based, and the rounding errors can cause snapping issues. 

See how we fix this in [gdsfactory](https://gdsfactory.github.io/gdsfactory/notebooks/08_pdk.html#PDK-decorator)

https://github.com/amccaugh/phidl/issues/140

nmz787-intel, 2023-02-10:
I've also been hit with rounding errors and had to questionably resort to fuzzy-matching. It certainly gives me pause about using this vs vendor tools (which have generally terrible APIs/ease-of-use, and I'd prefer to avoid).

joamatab, 2023-02-10:
https://gdsfactory.github.io/gdsfactory/notebooks/042_non-manhattan-router.html#Fix-Non-manhattan-connections

alibillalhammoud, 2023-07-07:
I also ran into significant issues with rounding errors and my project is completely Manhattan routing. I prefer using this tool, but these errors make it difficult. Can we at least get an option for more accurate floats, perhaps use long double by default or a more precise float from the boost library? Although not the robust/performant integer solution, it could at least prevent some of these errors.

joamatab, 2023-07-08:
@sebastian-goeldi
@tvt173

heitzmann, 2023-07-08:
> I also ran into significant issues with rounding errors and my project is completely Manhattan routing. I prefer using this tool, but these errors make it difficult. Can we at least get an option [...]

@alibillalhammoud Any chance you can give me an example of error you're facing? Maybe there are strategies to minimize them. As for moving towards integer coordinates, I've been looking into that but it is a very large refactor and I just don't have the time available to do it properly now.

smunaut, 2025-03-07:
Just to add my 2ct, I've also had precision errors ( I'm using python binding ) and the conversions back and forth between float64 and float32 end up messing things up

```
In [7]: np.float64(np.float32(np.float64(2300.6400000000003)))
Out[7]: np.float64(2300.639892578125)
```

Discrepency seem to pop-up mostly when a path gets converted to a polygon for instance and then its points don't actually match a neighboring native polygon because of such precision issue.