heitzmann/gdstk #142: clarify docs to state boolean sequence argument may be empty

nmz787-intel, 2023-01-18:
It isn't clear from the docs that an empty-sequence like this would work for boolean operations (i.e. merging all polygons into one, in one line):
```
import gdstk
lib = gdstk.Library()
cell = lib.new_cell("FIRST")
rect = gdstk.rectangle((0, 0), (2, 1))
rect2 = gdstk.rectangle((2, 0), (4, 1))
cell.add(rect)
cell.add(rect2)
len(cell.polygons)
# prints 2
or_output = gdstk.boolean(cell.polygons, [], "or")
len(or_output)
# prints 1
or_output[0].bounding_box()
# prints ((0.0, 0.0), (4.0, 1.0))
```

https://heitzmann.github.io/gdstk/geometry/gdstk.boolean.html#gdstk.boolean

heitzmann, 2023-01-20:
Done!

nmz787-intel, 2023-01-20:
Thanks, that speaks to my learning on the operand order with respect to the boolean operator, as this example demonstrates for `NOT`:

```

import gdstk
lib = gdstk.Library()
cell = lib.new_cell("FIRST")
rect = gdstk.rectangle((0, 0), (4, 4))
rect2 = gdstk.rectangle((3, 3), (4, 4))
rect3 = gdstk.rectangle((3, 0), (3, 1))
cell.add(rect)

# (one big polygon) - (smaller polygon) = (one polygon missing a chunk)
new_polygon = gdstk.boolean(cell.polygons, rect2, 'not', layer=1,datatype=10)
assert len(new_polygon)==1

# one polygon - (two others) = (one polygon missing two chunks)
new_polygon = gdstk.boolean(cell.polygons, [rect2, rect3], 'not', layer=1, datatype=10)
assert len(new_polygon) == 1

# (two small polygons) - (one larger, encompassing, polygon) = 0 polygons
new_polygon = gdstk.boolean([rect2, rect3], cell.polygons, 'not', layer=1, datatype=10)
assert len(new_polygon) == 0

```

closed 2023-04-01 by heitzmann