heitzmann/gdstk #143: Latest gdstk version whl to compatible with a specific numpy version

phylove01, 2023-01-28:
Hi, heizmann,

I am trying to install gdstk into my offline environment. I succeeded in installing latest version whl file(0.9.35) through pip, but I have an error message "RuntimeError: module compiled against API version 0x10 but this version of numpy is 0xe". Since, I have to use a specific version of numpy(1.20.1) due to the other libraries, could you share the latest version whl file to compatible with numpy 1.20.1? I tried all the other source files, gdstk-*.tar.gz on pypi, but I failed to install those through pip and setuptools because of the other dependency in my offline environment. Please help me to use gdstk!

joamatab, 2023-01-31:
Hi Lucas,

gdstk is failing in colab
[image]

@proppy
@heitzmann 
@lucas-flexcompute

proppy, 2023-01-31:
one possible workaround that we used for https://github.com/chipsalliance/silicon-notebooks/issues/30, is to do:
```
python -m pip install --no-binary gdstk gdstk
```
to force recompiling gdstk against the current environment.

joamatab, 2023-02-10:
Hi Lucas,

most users using colab are not able to install gdstk

I also tried upgrading numpy before but it does not worl

[image]

heitzmann, 2023-02-11:
I've just found out this package: https://pypi.org/project/oldest-supported-numpy/ which might be what we need to ensure the most compatible version of numpy possible, instead of using the latest available in each CI image. I'll try to set it up in the next few days to see how it goes. Maybe it will solve this issue in a more user-friendly way…

heitzmann, 2023-02-12:
Looks like it works now on release 0.9.37.

[image]

joamatab, 2023-02-12:
Yay! thank you Lucas!

works great on colab now!

[image]

closed 2023-02-12 by joamatab