heitzmann/gdstk #146: Parallel design creation

ThomasAuzinger, 2023-02-07:
Hi,

I am wondering if significant performance gains can be achieved by creating designs in parallel on multiple CPUs.
Let's assume I have 100 CPUs at my disposal and a design that consists of a large number of references - say 10^7. Is there a way to build a library significantly faster than with 1 CPU?

There seem to be possibilities to run different loops in parallel but it's not clear to me if `gdstk` would support this (in the sense that it provides an actual speedup and does not suffer from race conditions).

It would be great if someone with a deeper insight into `gdstk` could weigh in here.

Thanks and all the best,
Thomas

tvt173, 2023-02-07:
Hi @ThomasAuzinger , as always with parallelization, I think it depends on how independent the different calculations are. If we assume the best case (for parallelization) that all 10^7 cells are unique with no hierarchy, there should be nothing stopping you from using multiprocessing to create each cell in parallel and compose them at the end.

In reality you probably have some amount of interdependencies, hierarchy, and repetition, but you could probably make some sort of DAG (i.e. with Dask) to compute everything as efficiently as possible.

This assumes that you are trying to parallelize at a high level, in python. You could also do _multithreading_ of individual operations at the gdstk level, which would be very different.

Also note that it is highly efficient to create arrays and repetitions. If you are simply placing the same cell multiple times, you shouldn't need to introduce any form of parallelization.

@heitzmann may have addiitonal thoughts...

ThomasAuzinger, 2023-02-07:
Hi @tvt173 , thanks for the quick answer.

Creating the cells in parallel and composing them, in the end, sounds like the things that I want to do. Sadly, I do not have simple repetitions that can be done via references.
What I saw from the documentation are the commands `gdstk.Cell(...)` to create cells and `gdstk.Library.add(...)` to add them to the library. However, it seems to me that while `gdstk.Cell(...)` could be parallelized, `gdstk.Library.add(...)` still needs to run over all 10^7 cells in a serial manner; thus dropping the performance close to the unparallelized version.
Is this a correct interpretation?
If yes, should it be done differently (e.g., with different commands)?
If no, why is `gdstk.Library.add(...)` faster?

Sorry for the many questions but I am missing some central insight to understand the parallelization potential.

tvt173, 2023-02-07:
indeed, I would create subcells on separate processes and add them to the library in the main process. adding to the library should be a relatively cheap operation, so the scaling should still be good. remember, you can add cells into the main library on the main thread in parallel with the computation of the subcells on the worker threads, so I wouldn't expect the performance hit to be too bad. and for most applications, i would expect a more complicated, hierarchical graph task anyways (meaning there would be no single task handling 10^7 adds)

you may get serialization issues when trying to copy the cells between processes. but i find that when using dask for parallelization, i get less serialization issues than with plain multiprocessing. getting around those serialization issues is a more advanced topic 😉 

i've also been interested to try this out, so i'd be interested to know how it goes if you do attempt this. I usually use gdsfactory (which uses gdstk as the backend). it has caching capabilities that usually make building of large masks pretty fast without parallelization, so I've never had enough necessity to try.

heitzmann, 2023-02-08:
Creating the cells in separate processes should be indeed straightforward. Adding them to a single library has to be done in sync because the underlying array that holds the cell references is not thread-safe. That said, adding to the library is extremely cheap, so, as @tvt173 mentioned, it should not be a bottleneck (just use any sort of lock and you should be good to go). I expect the worst part to be saving the layout to disk. That's where no parallelization will help if you write the whole file at once. A better alternative might be to have a shared [`GdsWriter`](https://heitzmann.github.io/gdstk/library/gdstk.GdsWriter.html) between all processes and use it to output directly to the disk (again, with proper synchronization).

tvt173, 2023-02-08:
thinking about this further, you could also write individual cells to disk in each worker thread and compose later, loading as raw cells. should avert any serialization issues and also be extremely fast I/O, right?

heitzmann, 2023-02-08:
Yes, at some point, disk access becomes the limiting factor. Your suggestion might approach that limit!

closed 2023-06-25 by heitzmann