heitzmann/gdstk #147: `get_polygons` gives incorrect result when datatype omitted

nmz787-intel, 2023-02-10:
Here's a script that should demonstrate when `get_polygons` is constrained with `layer` but not `datatype`, it incorrectly returns all polygons in the cell, when the expectation is only a single polygon would be returned. Issue does not happen when both arguments are supplied. 

Version `0.9.35`.
Happens with GDS or OAS output/input.

```
import gdstk
import decimal
unit_value = 1e-6 #micrometer is the unit everyone passes data around with
precision = decimal.Decimal(unit_value) / decimal.Decimal(40000)
reference_dependency = 'my_top_level_cell1'

lib = gdstk.Library(unit=unit_value, precision=precision)
cell = lib.new_cell(reference_dependency)
cell.add(gdstk.rectangle((0, -10), (150, 10), layer=1, datatype=1))
cell.add(gdstk.rectangle((0, -10), (150, -9.5), layer=10000, datatype=0))
lib.write_oas('my_left_edge_case.oas')
lib.write_gds('my_left_edge_case.gds')

for l in [gdstk.read_oas('my_left_edge_case.oas'), gdstk.read_gds('my_left_edge_case.gds')]:
	print('checking file')
	len1 = (len(l.cells[0].get_polygons(layer=10000)))
	len2 = (len(l.cells[0].get_polygons(layer=10000, datatype=0)))
	len3 = (len([p for p in l.cells[0].polygons if p.layer==10000]))
	print(len1)  # prints 2
	print(len2)  # prints 1
	print(len3)  # prints 1
```

heitzmann, 2023-02-10:
The result is not incorrect. There's a note specifically documenting the [filtering behavior](https://heitzmann.github.io/gdstk/library/gdstk.Cell.html#gdstk.Cell.get_polygons)

The reason for that is performance. Filtering on a single value (layer or datatype) would not more efficient than simply doing directly:

```python
polys = [p for p in cell.get_polygons() if p.layer == my_layer]
```

nmz787-intel, 2023-02-10:
The documentation is not clear on both arguments being required. If a single item is ignored, raising an `Exception` seems better than silently ignoring and placing the burden on the user. The way I read the docs, the code should filter based on my input `layer==10000` and datatype==None would yield 0 polygons (as all my polygons have a non-NoneType datatype... as do /all/ polygons, I believe).

For me it isn't about efficiency of filtering, but efficiency of writing clean code (convenience of not having to copy-paste that list-comprehension around since the idea is used often).

Now, double-checking, I see you're referring to the NOTE... not the actual argument description/help.
I don't think I ever read this API section before, rather found `get_polygons` through reading old issues here.

Again, I think if both args are required, if only one is specified, it's safer to `raise Exception`.

Thanks for the clarification... now to search through my code for potentially-incorrect usage of `get_polygons` :O

heitzmann, 2023-02-11:
I see your point… I definitely like the idea of raising an exception (or at least a warning) if only one of the arguments is given, and that's easy to implement.
Adding the possibility of filtering by a single value can certainly be done, but I'm very short on time right now, so it will have to wait at least a couple of weeks.

closed 2023-06-25 by heitzmann