heitzmann/gdstk #148: How to merge all polygons in a list

inekash, 2023-02-12:
When creating a list of polygons and merging them 

p1 = gdstk.rectangle((-1, -1), (1, 1))
p2 = gdstk.rectangle((-0.5, -0.5), (1.5, 1.5))
p3 = gdstk.rectangle((0, 0), (2, 2))

lst = [p1, p2, p3]

PLGN = gdstk.boolean(lst, None, "or")

The following error appears:

"Argument operand2 must be a Polygon, FlexPath, RobustPath, References. It can also be a sequence where each item is one of those or a sequence of points."

gnawhleinad, 2023-02-12:
> The following error appears:
> 
> "Argument operand2 must be a Polygon, FlexPath, RobustPath, References. It can also be a sequence where each item is one of those or a sequence of points."

@inekash, it's as the error suggests and `operand2` is not a valid input.

You could try instead:

```
import gdstk
>>> p1 = gdstk.rectangle((-1, -1), (1, 1))
>>> p2 = gdstk.rectangle((-0.5, -0.5), (1.5, 1.5))
>>> p3 = gdstk.rectangle((0, 0), (2, 2))
>>> 
>>> lst = [p1, p2, p3]
>>> 
>>> PLGN = gdstk.boolean(lst, [], "or")
>>> gdstk.boolean(lst, [], "or")
[<gdstk.Polygon object at 0x102e185f0>]
```

frankxxu, 2023-02-13:
You can also try:
...
>>> gdstk.boolean(lst[0], lst[1:], "or")
[<gdstk.Polygon object at 0x7f0554ac2510>]

closed 2023-04-01 by heitzmann