heitzmann/gdstk #151: Problems with FlexPath.arc

ustclyh, 2023-02-22:
Hello,

I notice that, for the arc generated with `FlexPath.arc`, the two ends are not strictly perpendicular to the initial and final path direction (in the following example, the two ends are not strictly parallel to x-axis). And I also check the coordinate with KLayout, there is no point with its y-coordinate being accurately zero. 

In this way, when I want to add two FlexPaths separately, for example, a segment and an arc on different layers, their ends cannot join perfectly with each other. 

I don't know if it is a bug or there are more considerations in the realization of `FlexPath.arc`. 

gdstk version: 0.9.37, install with conda

```python
import gdstk
import numpy as np

lib = gdstk.Library()
cell = gdstk.Cell('TEST')

path = gdstk.FlexPath([0, 0], 1)
path.arc(10, np.pi, 0)  # generate an arc from pi to 0 with radius 10
cell.add(path)

lib.add(cell)
lib.write_gds('test.gds')
```

heitzmann, 2023-03-01:
@ustclyh Sorry for taking so long to reply, but I've been quite busy lately.

This is a known issue that has 3 workarounds: either start and finish your path with a straight section, use `RobustPath`, or use the `gdstk.ellipse` function to draw an isolated arc. The problem is that `FlexPath` is constructed based on the path and its numerical derivatives, and the derivative will deviate at the ends of isolated arcs due to limited range for evaluating the derivative. `RobustPath` and `ellipse` have analytical derivatives.
```python
import gdstk
import numpy as np

lib = gdstk.Library()
cell = gdstk.Cell("TEST")

path = gdstk.FlexPath((0, 0), 1)
path.arc(10, np.pi, 0)
cell.add(path)

path = gdstk.FlexPath((0, -1), 1, layer=1)
path.segment((0, 1), relative=True)
path.arc(10, np.pi, 0)
path.segment((0, -1), relative=True)
cell.add(path)

path = gdstk.RobustPath((0, 0), 1, layer=2)
path.arc(10, np.pi, 0)
cell.add(path)

arc = gdstk.ellipse(
    (10, 0), radius=10.5, inner_radius=9.5, initial_angle=0, final_angle=np.pi, layer=3
)
cell.add(arc)

lib.add(cell)
lib.write_gds("test.gds")
```

ustclyh, 2023-03-02:
> @ustclyh Sorry for taking so long to reply, but I've been quite busy lately.
>
> This is a known issue that has 3 workarounds: either start and finish your path with a straight section, use [...]

Many thanks for the helpful reminder! Actually I also find `FlexPath.parametric` has the same issue, and now I will turn to `RobustPath` :)

closed 2023-03-06 by ustclyh