heitzmann/gdstk #155: cell with references does not have bounding_box unless saved & reloaded

nmz787-intel, 2023-03-08:
```
import gdstk
import decimal
import os

decimal.getcontext().prec=9+3 # full picometer resolution
unit_value = 1e-6
precision = (decimal.Decimal(unit_value) / decimal.Decimal(1000)) / decimal.Decimal(40)

lib = gdstk.Library(unit=unit_value, precision=precision)

cell1 = lib.new_cell('cell1')
cell1.add(gdstk.rectangle((0, 0), (10, 10)))
# lib.write_oas(os.path.join(this_files_dir, 'cell1.oas'))
cell2 = lib.new_cell('cell2')
cell2.add(gdstk.rectangle((0, 0), (5, 5)))
# lib.write_oas(os.path.join(this_files_dir, 'cell2.oas'))
cell3 = lib.new_cell('cell3')
cell3.add(gdstk.rectangle((0, 0), (5, 5)))
# lib.write_oas(os.path.join(this_files_dir, 'cell3.oas'))

container = lib.new_cell('container')

container.add(gdstk.Reference('cell1'))
container.add(gdstk.Reference('cell1', origin=(10,0)))
container.add(gdstk.Reference('cell1', origin=(20,0)))

container.add(gdstk.Reference('cell2', origin=(0,10)))
container.add(gdstk.Reference('cell3', origin=(0,15)))

container.add(gdstk.Reference('cell3', origin=(25,10)))
container.add(gdstk.Reference('cell2', origin=(25,15)))
print(f'writing my_reference_test oas')
lib.write_oas('my_reference_test.oas')
print(container.bounding_box())
print(len(container.polygons), len(container.references),
      len(container.dependencies(True)))

try:
	ll,ur = container.bounding_box()
except TypeError as e:
	print(e)
	print('\n\n ^ that error seems weird ^')

try:
	container.flatten()
	print(len(container.polygons), len(container.references),
      len(container.dependencies(True)))
	ll,ur = container.bounding_box()
except TypeError as e:
	print(e)
	print('\n\n ^ that 2nd error seems weird ^')
	lib = gdstk.read_oas('my_reference_test.oas')
	container = [c for c in lib.cells if c.name=='container'][0]
	ll,ur = container.bounding_box()
print('now OK')
w = ur[0] - ll[0]
h = ur[1] - ll[1]
print(f'w should be 30, gdstk says ({w})... h should be 20, gdstk says {h}')
assert(w==30)
assert(h==20)
```

heitzmann, 2023-03-08:
The problem is that you're adding references by name. There is no way to link the name to the actual `Cell` object (we don't check naming conflicts during creation of the library, nor keep a dictionary of all created cells, because that could easily lead to inconsistencies). To make sure bounding box accounts for references, add them from the cell object:

```python
container.add(gdstk.Reference(cell1))    # do this
container.add(gdstk.Reference("cell1"))  # not this
```

nmz787-intel, 2023-03-08:
the docs specify a string is OK:
https://heitzmann.github.io/gdstk/library/gdstk.Reference.html
`cell (gdstk.Cell, gdstk.RawCell, str) – referenced cell.`

nmz787-intel, 2023-03-08:
the note is also confusing, as the operation works after reloading the file from disk
`If cell is a str (cell name), operations on the referenced cell are not performed, such as computing the bounding box.`

nmz787-intel, 2023-03-08:
Can you clarify when adding a reference by name is acceptable and won't produce unexpected (maybe for the novice/ignorant user) behavior?

heitzmann, 2023-03-08:
> the note is also confusing, as the operation works after reloading the file from disk

Because when loading from disk, the internal loader uses the objects, not the names, if the referenced cell with the corresponding name is present.

> Can you clarify when adding a reference by name is acceptable and won't produce unexpected (maybe for the novice/ignorant user) behavior?

It is always better to reference by object. Only use the name if you don't have the cell.

nmz787-intel, 2023-03-08:
>we don't check naming conflicts during creation of the library

this was another issue I ran into... which results in OAS files that are unreadable. It would be great if you could prevent that one from occurring. I ended up keeping a dictionary of the cellnames present in a lib, so I wouldn't add them multiple times (when adding references, I always add the actual cell too, because the few CAD viewers I know how to use don't look for references in files on disk... so any references need to be in the actual library itself)

heitzmann, 2023-03-20:
> I ended up keeping a dictionary of the cellnames present in a lib, so I wouldn't add them multiple times (when adding references, I always add the actual cell too, because the few CAD viewers I know [...]

That's exactly what you're supposed to do! It's hard to do that internally without ending up with performance problems. Imagine the user decides to build a library and them, at some point, renames a cell. We'd have to go through the whole library checking for duplicates. Now, if what we want to do is load a large library ans rename all cells, that becomes an n² problem. Now we have to add some sort of control flag to the library telling it not to perform name checks, which increases the complexity of the interface. That's why in these case I usually opt for the simpler solution, which might require more effort from the user, but leaves them in control, so that they can optimize performance for their use case. That's why `gdstk` is a low-level library. We try not to do any magic behind the scenes.

closed 2023-04-01 by heitzmann