heitzmann/gdstk #158: Filling cell with polygons avoiding overlaps to other layers

albe-jj, 2023-03-25:
I am currently using positioning filtering to place crosses around my design. 
I would like to have the crosses distributed as uniformly as possible but with position filtering I obtain the following
[image]

Is there a more intelligent way to do this? like a fill tool where you can fill certain areas with this crosses making sure they do not overlap with the design but that they fill the space if there is no cross in the radius of say 5 um?

joamatab, 2023-03-25:
Hi Alberto,

we have fill in gdsfactory, which is build on top of gdstk [see](https://gdsfactory.github.io/gdsfactory/notebooks/04_components_geometry.html#dummy-fill-tiling)

For filling big regions we are building a similar functionality using klayout fill tool
https://github.com/gdsfactory/gdsfactory/issues/1465

heitzmann, 2023-03-28:
Hi @albe-jj, did you take a look at the second example in the section [Geometry Filtering](https://heitzmann.github.io/gdstk/how-tos.html#geometry-filtering) in the docs? That's a more or less straightforward solution that doesn't require boolean operations. Anything fancier might require specialized algorithms, for example, defining the exact area you want filled through offsetting + boolean operations, simplifying the polygonal boundaries, extracting a Delaunay triangulation and using its centers for the fill placement within a certain size constraint, but that sounds like a lot of overkill depending on your use case.

closed 2023-06-25 by heitzmann