heitzmann/gdstk #159: FlexPath vertical and horizontal segments in different layers and datatypes

frankxxu, 2023-03-26:
Is there a way to make vertical and horizontal segments in different layers and datatypes even drop vias?

heitzmann, 2023-03-28:
I'm not sure I understand your question, @frankxxu. Can you maybe show an example of what you want?

frankxxu, 2023-03-31:
[image]
Left path can be created by gdstk.FlexPath.  Is there a way to create path on the right side using gdstk.FlexPath?

heitzmann, 2023-04-01:
Not directly, no. But I imagine it wouldn't be difficult to write a function that could automate this process for you if you need to use it repeatedly. If you already have the list of vertices to go through, you could loop through it, check whether the section you're on is vertical or horizontal and create the corresponding FlexPath for that section, then return all paths.

frankxxu, 2023-04-01:
Thanks for clarification.  I just want to make sure I don't miss any FlexPath features.  For 16nm and below, metals in vertical and horizontal directions are always different.

heitzmann, 2023-04-01:
I see, that's good to know! Maybe in the future we can include a function with this type of profile...