heitzmann/gdstk #162: Polygon.fracture cannot produce rectangles

nmz787-intel, 2023-04-03:
`maxpoints` arg minimum is 5.. which implies you cannot fracture a manhattan polygon into rectangles (rectangular polygons). Is there a quick workaround (I am failing to see any way forward with the current API, other than to stop drawing non-rectangular shapes from the get-go)?

heitzmann, 2023-04-08:
There's no current way in gdstk to transform all geometry into rectangles. But if you only use Manhattan geometry, it should be fairly simple to traverse all your elements and use [`slice`](https://heitzmann.github.io/gdstk/geometry/gdstk.slice.html#gdstk.slice) on every concave vertex you find (maybe iteratively, so you don't overdo it).

closed 2023-06-25 by heitzmann

nmz787-intel, 2023-09-14:
I tried, unsuccessfully, to replace the `points` in [this example](https://github.com/heitzmann/gdstk/issues/24#issuecomment-776624764) with those from my [current testcase](https://github.com/heitzmann/gdstk/issues/198), but it also fails to produce any rectangular shapes. Looking at the docs for `slice` I was unsure how to know where cut positions would be to supply `position (number or sequence) – Cut positions.`, but then I recalled asking about this before (this ticket).

I found some code that calculates a cross-product to determine concave, do you think it is OK in terms of computational efficiency?

```
import gdstk
t = gdstk.Library()
cell = t.new_cell("random")

def find_concave_vertices(points, coordinate='x'):
    concave_vertices = []
    n = len(points)

    for i in range(n):
        # Get the three consecutive points (p1, p2, p3)
        p1 = points[i]
        p2 = points[(i + 1) % n]
        p3 = points[(i + 2) % n]

        # Calculate vectors p1 -> p2 and p2 -> p3
        vec1 = (p2[0] - p1[0], p2[1] - p1[1])
        vec2 = (p3[0] - p2[0], p3[1] - p2[1])

        # Calculate the cross product of vec1 and vec2
        cross_product = vec1[0] * vec2[1] - vec1[1] * vec2[0]

        # Check the sign of the cross product
        if cross_product < 0:
            if coordinate == 'x':
                concave_vertices.append(p2[0])
            elif coordinate == 'y':
                concave_vertices.append(p2[1])
            else:
                raise ValueError("Invalid argument for 'coordinate'. Choose either 'x' or 'y'.")

    return list(set(concave_vertices))

polygon = gdstk.Polygon([[93140., 48500.],
 [50140., 48500.],
 [50140., 77500.],
 [93140., 77500.],
 [93140., 92500.],
 [32140., 92500.],
 [32140., 33500.],
 [93140., 33500.]])
cuts = find_concave_vertices(polygon.points, coordinate='x')
print(cuts)
slices = gdstk.slice(polygon, cuts, "x")
for s in slices:
    cell.add(*s)

# include the original in a different layer just to check
cell.add(polygon)
polygon.layer = 1

t.write_gds("cross_product_concave_point_detection_slicing.gds")
```

it does seem to solve the issue :)