heitzmann/gdstk #164: gdstk.text -> How can I change the texttype ?

EnesBae, 2023-04-07:
T_T

heitzmann, 2023-04-08:
@EnesBae The [`text`](https://heitzmann.github.io/gdstk/geometry/gdstk.text.html#gdstk.text) function creates text as *Polygons*, so they have `datatype`, not `texttype`. If you want to create textual elements, you're looking for [`Label`](https://heitzmann.github.io/gdstk/library/gdstk.Label.html#gdstk.Label). Those will have a `texttype` you can set.

EnesBae, 2023-04-09:
Is there a method in the Label function that can generate polygons like the Text function? I want to create polygons that are generated in a different texttype than the default texttype of the Text function.

thanks.

heitzmann, 2023-04-09:
To be clear, polygons don't have a `texttype` property, their equivalent is `datatype`, and, if you want to set those, it's a simple assignment:
```python
polygon_list = gdstk.text("text string", (0, 0), datatype=1)
# Change the datatype of the first polygon
polygon_list[0]. datatype = 2
```

heitzmann, 2023-04-09:
To be clear, polygons don't have a `texttype` property, their equivalent is `datatype`, and, if you want to set those, it's a simple assignment:
```python
polygon_list = gdstk.text("text string", (0, 0), datatype=1)
# Change the datatype of the first polygon
polygon_list[0]. datatype = 2
```

The `Label` class represents labels, not polygongs, so it cannot generate geometry by itself. You could write a function that take in a label and uses the `text` function to generate a polygonal representation of that label with any datatype you want.

EnesBae, 2023-04-09:
The current font used in the text function is not suitable for my field of work. I want to generate polygons with a simpler font that doesn't have diagonal patterns using the text function. Is there any way to do this?
thanks.

heitzmann, 2023-04-09:
The current font has no diagonal lines. But if you need something specific, there's an example on [how to use any system font](https://heitzmann.github.io/gdstk/how-tos.html#system-fonts).

Beyond that, you'd have to write your own function to convert text into polygons.

EnesBae, 2023-04-09:
> The current font has no diagonal lines. But if you need something specific, there's an example on [how to use any system font](https://heitzmann.github.io/gdstk/how-tos.html#system-fonts).
> 
> Beyond that, you'd have to write your own function to convert text into polygons.

Perfect !! Thank you so much !!

closed 2023-06-25 by heitzmann