heitzmann/gdstk #165: Increase max_points in boolean operation

larchen, 2023-04-07:
In gdspy there was the option to specify `max_points` to a larger value in the boolean, offset, etc functions. Would it be possible to have this feature in gdstk as well?

Alternatively is there a way to reverse a fracturing operation and combine a list of polygons into a single polygon?

tvt173, 2023-04-07:
I believe in gdstk all the fracturing happens at write-time, so there is no need to specify fracturing parameters during these operations. Likewise, if you run an `or` operation on all the shapes in the layer you want to merge, you should be able to "reverse the fracturing operation". You should get the union of shapes back, as explained below.
[image]

heitzmann, 2023-04-08:
@larchen Troy is correct, in gdstk those operations do not fracture your polygons, that's why `max_points` doesn't exist there.

larchen, 2023-04-09:
Thanks for the clarification -- I missed the `max_points` argument for `write_gds` when I was looking at the documentation earlier.

closed 2023-04-09 by larchen