heitzmann/gdstk #166: Reference.repetition sometimes outputs a strange value !!

bsslv, 2023-04-22:
Like the Title, sometimes "Refence.repetition" attribute output a strange value, 
like when correct result of the attribute is "10", the function Spit "0" with ignorance attitude.

PLZ check this situation..

heitzmann, 2023-04-22:
Can you give a reproducible test case? I have not seen this happen. Please note that if you use apply_repetition, its value will be changed to None, as explained in the documentation: https://heitzmann.github.io/gdstk/library/gdstk.Reference.html#gdstk.Reference.apply_repetition

bsslv, 2023-04-22:
Hm.. It s impossible to give you the test case,,, the oasis file is security file.

-. Im not use the "apply_repetition() you told me
-. If I explain the situation as best as possible, the hierarchy is 17 level, and at the bottom level, the function "Reference.repetition" work well, but not after the bottom level ; I mean the bottom level is not top_level()'s cell.

heitzmann, 2023-04-22:
Can you show the script you're using to load the file and the commands and replies you see without showing the file contents?

bsslv, 2023-04-24:
You can test using this script.

THX

########################################################
########################################################
########################################################
import gdstk
import numpy as np

lib = gdstk.Library()
# TOP Cell
cell_top = lib.new_cell("TOP")
cell_a = lib.new_cell("A")
cell_b = lib.new_cell("B")

# Make polygons of A level
poly_a = gdstk.Polygon([(0, 0), (1, 0), (1, 1), (0, 1)])
rep_a = gdstk.Repetition(columns=1, rows=3, spacing=(0, 2))
poly_a.repetition = rep_a

# Make polygons of B level
poly_b = gdstk.Polygon([(0.5, 0.5), (1.5, 0.5), (1.5, 1.5), (0.5, 1.5)])
rep_b = gdstk.Repetition(columns=4, rows=1, spacing=(2, 0))
poly_b.repetition = rep_b

# "A level" added to the cel_a 
cell_a.add(poly_a)
# "B level" added to the TOP Cell for reference on the TOP CELL
cell_b.add(poly_b)

# "A level" added to the cell_a for reference on the Cell a
cell_b.add(gdstk.Reference(cell_a)) 
# "B level" added to the TOP Cell for reference on the TOP CELL
cell_top.add(gdstk.Reference(cell_b)) # CellRef 대신 Reference 사용

############### Repetition Attribute TEST ##################
### Compare the repetition arrrubure with using reference and NOR ###
H1 = cell_top.references[0]
H1.cell.name
np.size(H1.repetition) # outout ; 4
np.size(poly_b.repetition) # outout ; 0

H2 = H1.cell.references[0]
H2.cell.name
np.size(H2.repetition) # outout ; 0
np.size(poly_a.repetition) # outout ; 3
########################################################
########################################################
########################################################

bsslv, 2023-04-24:
OK, I'll show you example code script after checking,PLZ give me some time

bsslv, 2023-04-26:
You can check the function of repetition attribute of cell address

##################################################
##################################################
##################################################
from gdstk import *
import numpy as np

lib = Library()

## TOP CELL
cell_top = lib.new_cell("TOP")
cell_A = lib.new_cell("A")
cell_B = lib.new_cell("B")

## Define the polygon for layers A 
polygon_A = Polygon( [(0, 0), (1, 0), (1, 1), (0, 1)] )
polygon_A.repetition = Repetition(columns=1, rows=3, spacing=(0,2))

## Define the polygon for layers B
polygon_B = Polygon( [(0, 0), (1, 0), (1, 1), (0, 1)] )
polygon_B.repetition = Repetition(columns=4, rows=1, spacing=(0,2))

## "A" layer added to the cell_A
cell_A.add(polygon_A)

## "B" layer added to the cell_B
cell_B.add(polygon_B)

# "cell_A" referenced on the "cell_B"
cell_B.add(Reference(cell_A))

# "cell_B" referenced on the "cell_top"
cell_top .add(Reference(cell_B))

#######################################
############ Hierarchy Structure ############
#######################################
###### Top ; cell_top
###### Second ; B, repetition size ; 4
###### Bottom ; A, repetition size ; 3
#######################################
#######################################

###### TEST ######
H1 = cell_top.references[0] 
print("H1 cell name ;", H1.cell.name)
print("H1 repetition size using repetition attribute ;",np.size(H1.repetition)) ## output is "0"
print("H1 repetition real size ;",np.size(polygon_B.repetition)) ## output is "4"

print("\n")

H2 = H1.cell.references[0] 
print("H2 cell name ;",H2.cell.name) ## output is "A"
print("H2 repetition size using repetition attribute ;",np.size(H2.repetition)) ## output is "0"
print("H2 repetition real size ;",np.size(polygon_A.repetition)) ## output is "3"

##################################################
##################################################
##################################################

heitzmann, 2023-04-29:
Hi @bsslv, now I understand the confusion. These are different things: `np.size(H1.repetition)` counts the number of repetitions of the `Reference` object, whereas `np.size(polygon_B.repetition)` is the number of repetitions of the `Polygon` object. If you want to create a repetition of references (reference array), then set the repetition value of the reference object:
```python
H1 = gdstk.Reference(cell_B)
H1.repetition = Repetition(columns=4, rows=1, spacing=(0,2))
top.add(H1)
```

bsslv, 2023-04-30:
Thx, 

yes, I know the method how to store the array information.
but, the H1.repetition don t have any information like below image
[image]
you can check

bsslv, 2023-04-30:
Thx your reply,

Then, How do I call the array information of the other level when I only know the bottom level.

My object is the calling the array information from second level to top level using bottom level.

bsslv, 2023-04-30:
Hi, first of all THX for your reply

I understood the difference between the number of repetitions of the Reference and of the Polygon.

Then, How do I get the Cell repetition information from the Cell address?

For example, Polygon A is added to the Cell_A, 
then, Cell_A is referenced on the Cell_B, and Cell_B is repeated twice in the row direction every (2,0) spacing.

I know polygon repetition method like below text.
=====
polygon_A = Polygon( [(0, 0), (1, 0), (1, 1), (0, 1)] )
polygon_A.repetition = Repetition(columns=4, rows=1, spacing=(0,2))
=====

I NEED TO KNOW How to check the information that repeats the cell itself.
Is there any other method other than "get_polygons()" method?

heitzmann, 2023-05-01:
From your example, you'd have something like this:
```python
polygon_A = Polygon( [(0, 0), (1, 0), (1, 1), (0, 1)] )
cell_A = gdstk.Cell("A")
cell_A.add(polygon_A)

ref_A = gdstk.Reference(cell_A)
cell_B = gdstk.Cell("B")
cell_B.add(ref_A)

ref_B = gdstk.Reference("B", columns=2, rows=1, spacing=(2, 0))
cell_top = gsdtk.Cell("TOP")
cell_top.add(ref_B)
```
In this case, to find the repetition count, you'll have to check `ref_B.repetition`.

bsslv, 2023-05-02:
Thank you bro. I solved what my confusion. :)

closed 2023-06-25 by heitzmann