heitzmann/gdstk #175: speed up remapping layers

joamatab, 2023-05-19:
In klayout remapping layers is quite fast

however gdstk remapping layers is pretty slow as it's done at the polygon level

how can we remap layers on gds at the read level?

heitzmann, 2023-05-24:
We could implement something like `Library.remap_tags(**tags_dict)` and `Cell.remap_tags(**tags_dict)` to remap over the whole library/cell. I'm not a fan of coupling it to `read_gds` or `read_oas`, as it would be useless for in-memory libraries.

I don't see how remapping during loading would faster than after the contents are loaded. In either case, for each element in the library, the layer is looked up in a hash table and remapped to the resulting value, if any. There might only be a caching difference, since on the second option we'd iterate over the library twice (loading, then remapping) but the total computational cost remains the same. Am I missing something?

joamatab, 2023-05-29:
Hi Lucas,

what's the recommended way to remap a layer in gdstk?

in gdsfactory we've been extracting all polygons, which is quite slow for big layouts

```python
   def remap_layers(
        self, layermap, include_labels: bool = True, include_paths: bool = True
    ) -> Component:
        """Returns a copy of the component with remapped layers.

        Args:
            layermap: Dictionary of values in format {layer_from : layer_to}.
            include_labels: Selects whether to move Labels along with polygons.
            include_paths: Selects whether to move Paths along with polygons.
        """
        component = self.copy()
        layermap = {_parse_layer(k): _parse_layer(v) for k, v in layermap.items()}

        all_D = list(component.get_dependencies(True))
        all_D.append(component)
        for D in all_D:
            for p in D.polygons:
                layer = (p.layer, p.datatype)
                if layer in layermap:
                    new_layer = layermap[layer]
                    p.layer = new_layer[0]
                    p.datatype = new_layer[1]
            if include_labels:
                for label in D.labels:
                    original_layer = (label.layer, label.texttype)
                    original_layer = _parse_layer(original_layer)
                    if original_layer in layermap:
                        new_layer = layermap[original_layer]
                        label.layer = new_layer[0]
                        label.texttype = new_layer[1]

            if include_paths:
                for path in D.paths:
                    new_layers = list(path.layers)
                    new_datatypes = list(path.datatypes)
                    for layer_number in range(len(new_layers)):
                        original_layer = _parse_layer(
                            (new_layers[layer_number], new_datatypes[layer_number])
                        )
                        if original_layer in layermap:
                            new_layer = layermap[original_layer]
                            new_layers[layer_number] = new_layer[0]
                            new_datatypes[layer_number] = new_layer[1]
                    path.set_layers(*new_layers)
                    path.set_datatypes(*new_datatypes)
        return component
```

https://gdsfactory.github.io/gdsfactory/_modules/gdsfactory/component.html#Component.__init__

heitzmann, 2023-05-29:
That's what I have in mind indeed. I'd might optimize a little the conversions, but that's the overall algorithm.
```python
for p in D.polygons:
    layer = (p.layer, p.datatype)
    new_layer = layermap.get(layer, None)
    if new_layer is not None:
        p.layer, p.datatype = new_layer
```
Implementing it in gdstk would only move the code from python to C, nothing more than that.

joamatab, 2023-05-29:
i think exposing slow for loops in C++ would speed things quite a bit :)

joamatab, 2023-06-07:
This is super fast in klayout BTW

@tvt173
@sebastian-goeldi

sebastian-goeldi, 2023-06-07:
You won't be able to beat klayout in this regard I think. That's the advantage of having the shapes not associated with layers. Since only the container or rather the cell needs to know the layer (index actually), all you logically have to do is `cell.shapes(layer_index_A).layer_index=layer_index_B`. That's also why it's the same speed whether on the layout or the cell, as for one you just replace the index and the other you replace layer number/datatype.

heitzmann, 2023-06-25:
I see, associating the layer to the cell would indeed make such changes trivial. This would be a large change in the inner works of the library, though, so I'm not sure it'll ever be implemented.

tvt173, 2023-06-26:
I would suggest picking the easy to reach fruits first 😄 

Can we go with your original suggestion to just implement the loops in c++ @heitzmann ? It may not give better performance than Klayout, but it will still be a huge improvement on what we have now.

heitzmann, 2023-07-08:
@joamatab @tvt173 I've added the remap method to `Library` and `Cell`. Let mo know if you find any problems with it.

joamatab, 2023-10-08:
I will make a release so we can test it out in gdsfactory remap

joamatab, 2023-10-09:
it worked great, thank you Lucas!

https://github.com/gdsfactory/gdsfactory/pull/2168

closed 2023-10-09 by joamatab