heitzmann/gdstk #178: Reference arrays from individual particle locations

arondahl, 2023-06-05:
Is there any way to create large arrays of circular particles with aperiodic locations? Creating individual references takes way too much memory, whereas creating an array reference with the gdstk.Reference(columns,rows,spacing) is really good but it seems to me that its only for rectangular periodicity.

Thank you in advance!

heitzmann, 2023-06-05:
The repetition property of references (or polygons) accepts irregular arrays. Take a look at [this how-to](https://heitzmann.github.io/gdstk/how-tos.html#repetitions). You can use the polygon repetition or a reference repetition. OASIS files supports this kind of irregular repetition directly (in both polygon and reference), GDSII will have to store each instance separately, so a reference might be preferable to reduce the file size.

heitzmann, 2023-06-14:
That's correct. GDSII arrays are limited to rectangular, periodic distributions only.

closed 2023-06-25 by heitzmann