heitzmann/gdstk #182: Remove Holes form Polygons

ethanVodka, 2023-06-20:
I tryto remove a hole from a polygon in a gds file and save it as a new gds file,
Is there any good way to do this?
I am using C++ for the gdstk library.

[image]

heitzmann, 2023-06-25:
Not directly. You can first subtract the polygon from its bounding box (or the convex hull to get fewer false positives). That will leave you with all holes and concave cutouts. The idea is then to merge the original polygon with the holes or (better, in my opinion) subtract the cutouts from a new bounding box (or convex hull, if you started with it).

Sorting the pieces into holes and cutouts is the tricky part. The way I can see it being possible is by checking if each piece intersects with the external boundary (bounding box or convex hull): offset the bounding box (convex hull) inwards by a small amount and subtract it from the original. That will give the external contour. Then, intersect each piece with this contour: the ones with any intersection are exterior cutouts.

ethanVodka, 2023-06-25:
Thank you very much for your reply.

poly1 ={ {0, 0}, {50, 0}, {50, 50}, {0, 50} }
poly2 = {{20, 20}, {20, 30}, {30, 30}, {30, 20}}

I am now able to split inside and outside using Clipper, I am thinking of merging inside and outside to generate the points[] below

points[] = { {0, 0}, {50, 0}, {50, 50}, {20, 20}, {20, 30}, {30, 30}, {30, 20}, {20, 20} };

Is it possible to realize 
poly1 + poly2 = points[]
 in gdstk cpp? I would be very happy if you could show us the actual cpp code with it.

Here is my code split polygons into outer and inner
``` 
	List<List<GPoint^>^>^ outer = gcnew List<List<GPoint^>^>();
	List<List<GPoint^>^>^ inner = gcnew List<List<GPoint^>^>();

	for each (List<GPoint^> ^ poly in GdsPolygons[DisplayLayer]->plygons_list)
	{
		if (Utils::Common::UClipper::IsClockwise(poly))
		{
			outer->Add(poly);
		}
		else
		{
			inner->Add(poly);
		}
	}
```

heitzmann, 2023-06-25:
I'm not sure I understand how you want to merge those points. The array you describe (`{ {0, 0}, {50, 0}, {50, 50}, {20, 20}, {20, 30}, {30, 30}, {30, 20}, {20, 20} }`) is not a valid polygon.

heitzmann, 2023-06-25:
By the way: the suggestion I gave comes from https://github.com/heitzmann/gdstk/issues/173, if you wanto to take a look.

ethanVodka, 2023-07-03:
hi @heitzmann , That issue was quite helpful. thank for your advice.

ethanVodka, 2023-07-03:
hi, @tvt173. I was able to find a library with similar functionality to shapely and will try it out.
Thank you too!

closed 2023-07-08 by heitzmann