heitzmann/gdstk #184: [Suggestion] Add typing stubs

rassouly, 2023-06-21:
One of the only drawback of gdstk compared to gdspy is that IDE support is much poorer. For example, VSCode doesn't provide autocompletion at all. MyPy's stubgen can automatically generate a stub file to fix the autocompletion issue but the stub file doesn't contain any of the type information stored in the docstrings.
I don't see a super easy way of converting the docstrings into typestubs but writting it by hand should be quite quick. 
Would you be interested in such a patch?

heitzmann, 2023-06-25:
Sure, if you can write it we can include it in the package!

rassouly, 2023-06-29:
This was done in #185.
Could you make a release so that my rest of my lab can get access to the binary directly (setting up the compiler on windows is pretty annoying)?

rassouly, 2023-07-17:
Closed by #185

closed 2023-07-17 by rassouly