heitzmann/gdstk #187: Symbol not found error when running gdsfactory

NiteshChauhan1, 2023-06-29:
Hi,

I just installed gdsfactory and it calls gusto, it is giving me following error:
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
Cell In[8], line 1
----> 1 import gdsfactory as gf
      2 from gdsfactory.generic_tech import get_generic_pdk
      4 gf.config.rich_output()

File ~/miniconda3/lib/python3.10/site-packages/gdsfactory/__init__.py:23
     21 import warnings
     22 from toolz import compose
---> 23 from gdsfactory.component_layout import Group
     24 from gdsfactory.path import Path
     27 # NOTE: import order matters. Only change the order if you know what you are doing

File ~/miniconda3/lib/python3.10/site-packages/gdsfactory/component_layout.py:13
     11 import shapely as sp
     12 import numpy as np
---> 13 from gdstk import Label as _Label
     14 from gdstk import Polygon
     15 from numpy import cos, pi, sin

ImportError: dlopen(/Users/niteshchauhan/miniconda3/lib/python3.10/site-packages/gdstk.cpython-310-darwin.so, 0x0002): symbol not found in flat namespace '__ZN5gdstk12error_loggerE'
----------------------------------------------------------------------------------------------------------------------------------

Any idea how to resolve this? Any help is much appreciated!

tvt173, 2023-06-29:
Hi @NiteshChauhan1 , how did you install gdstk?
I highly recommend installing via mamba (rather than pip). If you type `mamba list gdstk` at the terminal, it should indicate which was used

joamatab, 2023-07-01:
Hi

This happens on m1 MacBook 

You need to 

conda install gdstk 

Then 

Pip install gdsfactory

closed 2023-07-08 by heitzmann