heitzmann/gdstk #188: Calculating the bounding_box() of the Сell before and after the flatten()

FedoninMichael, 2023-07-04:
Hello! Sorry if I'm duplicating my problem in issues. When I have a Cell with only references. And I get the bounding_box() of that Cell. Example - (x0, y0)(x1, y1). After that, I call the flatten() method and get the other bounding_box() coordinates. Example - (x0, y0)(x2, y2). Why? I repeat, Cell only has references. Maybe the calculation of bounding_box happens up to a certain level of references?

heitzmann, 2023-07-06:
Can you share an example where this issue happens?

FedoninMichael, 2023-07-11:
Sorry, code is a big. I am presenting fragment of the code:
[image]
And here result:
[image]

heitzmann, 2023-08-06:
Sorry, but without being able to reproduce it locally, I can't see what the problem might be. The examples I try here all work.

FedoninMichael, 2023-09-07:
Sorry for a longtime answer.

`import numpy as np
import os
from gdstk import Polygon, Cell, Reference, GdsWriter

output_root = os.getcwd()

def One_cross(width, length):
    points = Polygon([
        (-length / 2, -width / 2),
        (-length / 2, width / 2),
        (-width / 2, width / 2),
        (-width / 2, length / 2),
        (width / 2, length / 2),
        (width / 2, width / 2),
        (length / 2, width / 2),
        (length / 2, -width / 2),
        (width / 2, -width / 2),
        (width / 2, -length / 2),
        (-width / 2, -length / 2),
        (-width / 2, -width / 2),
    ])
    cross = Cell("code_base_mark").add(points)
    return cross

def gen_cross(width, height, angle):
    cross = Cell("cross")
    cross.add(Reference(One_cross(width, height)))
    cross = cross.copy("cross", rotation=angle*np.pi/180)
    cross = Reference(cross, origin=(-cross.bounding_box()[0][0], -cross.bounding_box()[0][1]))
    return cross

vertical_n = 10
horizontal_n = 10
def generate(line_width, line_size, distance, angle):
    max_size_x = 7.0
    max_size_y = 4.0
    C = Cell("C")
    shift_y = 0
    for y in range(vertical_n):
        line = Cell("line")
        shift_x = 0
        xsize = 0
        for x in range(horizontal_n):
            d = Cell("d").add(gen_cross(line_width, line_size, angle))
            d = Reference(d, origin=(shift_x, shift_y))
            line.add(d)
            if (xsize + distance + (line.bounding_box()[1][0] - line.bounding_box()[0][0])) > max_size_x:
                break
            shift_x = (shift_x + (d.bounding_box()[1][0] - d.bounding_box()[0][0]) + distance)
            xsize = (d.bounding_box()[1][0] - d.bounding_box()[0][0])
        temp = Reference(line, origin=((line.bounding_box()[1][0] - line.bounding_box()[0][0])/2 +
                                      line.bounding_box()[0][0], (line.bounding_box()[1][1] - line.bounding_box(
                                       )[0][1])/2 + line.bounding_box()[0][1]))
        C.add(temp)
        if (C.bounding_box()[1][1] - C.bounding_box()[0][1]) + distance + (d.bounding_box()[1][1] -
                                                                           d.bounding_box()[0][1]) > max_size_y:
            break
        shift_y = shift_y + (d.bounding_box()[1][1] - d.bounding_box()[0][1]) + distance
    C = C.copy("Cell", translation=(-C.bounding_box()[0][0], -C.bounding_box()[0][1]))
    return C

def make_job(output_root, line_width, line_size, distance, angle):
    cell = generate(line_width, line_size, distance, angle)
    file_name = cell.name
    cell = cell.copy(cell.name, translation=(0, -cell.bounding_box()[0][1]))
    cell = cell.copy(cell.name, translation=(-cell.bounding_box()[0][0], 0))
    path = os.path.join(output_root, file_name + ".gds")
    print(cell.bounding_box())
    cell.flatten()
    print(cell.bounding_box())
    gds_writer = GdsWriter(path, f"{cell.name}", unit=10e-6, precision=10e-8)
    gds_writer.write(cell)
    gds_writer.close()

line_width = 0.33
line_size = 0.99
distance = 0.69
angle = 45.0
make_job(output_root, line_width, line_size, distance, angle)`

This script gets to me different bounding_box before and after flatten() in "def make_job".

FedoninMichael, 2023-09-08:
Sorry for a longtime answer.

```
import numpy as np
import os
from gdstk import Polygon, Cell, Reference, GdsWriter

output_root = os.getcwd()

def One_cross(width, length):
    points = Polygon([
        (-length / 2, -width / 2),
        (-length / 2, width / 2),
        (-width / 2, width / 2),
        (-width / 2, length / 2),
        (width / 2, length / 2),
        (width / 2, width / 2),
        (length / 2, width / 2),
        (length / 2, -width / 2),
        (width / 2, -width / 2),
        (width / 2, -length / 2),
        (-width / 2, -length / 2),
        (-width / 2, -width / 2),
    ])
    cross = Cell("code_base_mark").add(points)
    return cross

def gen_cross(width, height, angle):
    cross = Cell("cross")
    cross.add(Reference(One_cross(width, height)))
    cross = cross.copy("cross", rotation=angle*np.pi/180)
    cross = Reference(cross, origin=(-cross.bounding_box()[0][0], -cross.bounding_box()[0][1]))
    return cross

vertical_n = 10
horizontal_n = 10
def generate(line_width, line_size, distance, angle):
    max_size_x = 7.0
    max_size_y = 4.0
    C = Cell("C")
    shift_y = 0
    for y in range(vertical_n):
        line = Cell("line")
        shift_x = 0
        xsize = 0
        for x in range(horizontal_n):
            d = Cell("d").add(gen_cross(line_width, line_size, angle))
            d = Reference(d, origin=(shift_x, shift_y))
            line.add(d)
            if (xsize + distance + (line.bounding_box()[1][0] - line.bounding_box()[0][0])) > max_size_x:
                break
            shift_x = (shift_x + (d.bounding_box()[1][0] - d.bounding_box()[0][0]) + distance)
            xsize = (d.bounding_box()[1][0] - d.bounding_box()[0][0])
        temp = Reference(line, origin=((line.bounding_box()[1][0] - line.bounding_box()[0][0])/2 +
                                      line.bounding_box()[0][0], (line.bounding_box()[1][1] - line.bounding_box(
                                       )[0][1])/2 + line.bounding_box()[0][1]))
        C.add(temp)
        if (C.bounding_box()[1][1] - C.bounding_box()[0][1]) + distance + (d.bounding_box()[1][1] -
                                                                           d.bounding_box()[0][1]) > max_size_y:
            break
        shift_y = shift_y + (d.bounding_box()[1][1] - d.bounding_box()[0][1]) + distance
    C = C.copy("Cell", translation=(-C.bounding_box()[0][0], -C.bounding_box()[0][1]))
    return C

def make_job(output_root, line_width, line_size, distance, angle):
    cell = generate(line_width, line_size, distance, angle)
    file_name = cell.name
    cell = cell.copy(cell.name, translation=(0, -cell.bounding_box()[0][1]))
    cell = cell.copy(cell.name, translation=(-cell.bounding_box()[0][0], 0))
    path = os.path.join(output_root, file_name + ".gds")
    print(cell.bounding_box())
    cell.flatten()
    print(cell.bounding_box())
    gds_writer = GdsWriter(path, f"{cell.name}", unit=10e-6, precision=10e-8)
    gds_writer.write(cell)
    gds_writer.close()

line_width = 0.33
line_size = 0.99
distance = 0.69
angle = 45.0
make_job(output_root, line_width, line_size, distance, angle)
```
This script gets to me different bounding_box before and after flatten() in "def make_job".