heitzmann/gdstk #189: The method of using filter of read_gds

bsslv, 2023-07-09:
Hello. I want to use the filter parameter of read_gds.

Is this code correct? read_gds(infile_str, filter={(1, 0), (2, 0)}) 
I tested this code, but a specific cell only made polygons of layer 1, 
BUT I know this cell has polygons of layer 2.

How can I fix the code?

I already tested “read_gds(infile_str, filter=list({(1, 0), (2, 0)}))”

THANKS

heitzmann, 2023-08-06:
Your calling argument is incorrect. You need to pass `filter` as an iterable of tuples, such as `{(1, 0), (2, 0)}` or `[(1, 0), (2, 0)]`, but you are passing a list with a set of tuples.

heitzmann, 2023-08-06:
(Misread your question). The first call is correct `read_gds(infile_str, filter={(1, 0), (2, 0)})`. Can you share an example with the problem so that I can investigate?