heitzmann/gdstk #191: Error with Boolean or operation on a Repetition

the-dave-power, 2023-07-14:
Hi @heitzmann ,

I am trying to do a boolean or operation on a Repetition I am not having much success. 

I looked at your how-to (https://heitzmann.github.io/gdstk/how-tos.html) and tried to follow it. But I am getting to following error: 

  File "/home/dpower/PycharmProjects/pythonProject/wafer_map_yh_hour_glass_checker.py", line 31, in <module>
    checker = gdstk.boolean(rect_array_odd,rect_array_even,"or",layer=3, datatype=0)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Unable to parse item 3 from sequence operand1.

Any chance you can take a look at this?

Thanks,
Dave.

code:
import gdstk

cell = gdstk.Cell('Top')

wafer_circum = 300
wafer_diam = wafer_circum/2
triangle_width = 180
rect_array_size = 0.030
#n_rows_cols = int(wafer_diam//0.030)
n_rows_cols = 2

wafer = gdstk.ellipse((0, 0), wafer_diam, tolerance=0.01,layer=1, datatype=0)

top_triangle =gdstk.Polygon([(0,0),( (triangle_width/2)   ,wafer_diam   ),( (triangle_width/-2),    wafer_diam  )], layer=1, datatype=0)
bot_triangle =gdstk.Polygon([(0,0),( (triangle_width/2),(-1*wafer_diam) ),( (triangle_width/-2),(-1*wafer_diam) )], layer=1, datatype=0)
or_triangle  = gdstk.boolean(top_triangle,bot_triangle,"or",layer=2, datatype=0)

rect_array_odd =  gdstk.rectangle(( (-1*wafer_diam), (-1*wafer_diam)), ((-1*wafer_diam+rect_array_size), (-1*wafer_diam+rect_array_size)), layer=3, datatype=0)
rect_array_even = gdstk.rectangle(( (-1*wafer_diam+rect_array_size),   (-1*wafer_diam+rect_array_size)), ((-1*wafer_diam+(rect_array_size*2)), (-1*wafer_diam+(rect_array_size*2))), layer=3, datatype=0)
rect_array_odd.repetition  = gdstk.Repetition(columns=n_rows_cols,rows=n_rows_cols,spacing=((rect_array_size*2),(rect_array_size*2)))
rect_array_even.repetition = gdstk.Repetition(columns=n_rows_cols,rows=n_rows_cols,spacing=((rect_array_size*2),(rect_array_size*2)))

# https://heitzmann.github.io/gdstk/how-tos.html
rect_array_odd  = rect_array_odd.apply_repetition()
rect_array_even = rect_array_even.apply_repetition()

rect_array_odd.append(rect_array_odd)
rect_array_even.append(rect_array_even)

checker = gdstk.boolean(rect_array_odd,rect_array_even,"or",layer=3, datatype=0)

yh_image_1   = gdstk.boolean(or_triangle,wafer,"and",layer=0, datatype=1)
yh_image_1   = gdstk.boolean(yh_image_1,wafer,"and",layer=0, datatype=1)

cell.add(wafer)

cell.add(*or_triangle)
cell.add(*yh_image)

lib = gdstk.Library(unit=0.001)
lib.add(cell)

lib.write_oas('/home/dpower/layout/yh_wafer_maps/hour_glass_checker.oas'

heitzmann, 2023-07-15:
The problem is that you are appending a list to itself in these lines:
```python
rect_array_odd = rect_array_odd.apply_repetition()
...
rect_array_odd.append(rect_array_odd)
```
Maybe try:
```python
rect_odd = gdstk.rectangle((-wafer_diam, -wafer_diam), (-wafer_diam+rect_array_size, -wafer_diam+rect_array_size), layer=3, datatype=0)
rect_odd.repetition = gdstk.Repetition(columns=n_rows_cols, rows=n_rows_cols, spacing=(rect_array_size*2, rect_array_size*2))
rect_array_odd = rect_odd.apply_repetition()
rect_array_odd.append(rect_odd)
```

By the way, you can use 3 backticks to format your code:
```
```python
# python code goes here
x = 1
y = 2 * x
``` 
```

the-dave-power, 2023-07-16:
@heitzmann 

thanks for the suggestion. I took the append statement. The code executes but does not finish running.

any ideas? 

Code:

[wafer_map_yh_hour_glass_checker.py.txt](https://github.com/heitzmann/gdstk/files/12064910/wafer_map_yh_hour_glass_checker.py.txt)

the-dave-power, 2023-07-17:
Hi @heitzmann 

Sorry read your initial response too quickly.  The below code works well using your suggestion :) 

  [wafer_map_yh_hour_glass_checker.py.txt](https://github.com/heitzmann/gdstk/files/12073007/wafer_map_yh_hour_glass_checker.py.txt)

My last issue is that I am trying to  make a array of 5k x 5k rectangles. It takes a couple of hours to run. Is there a way to improve the performance?  

Thanks,
Dave.

Thanks,
Dave.

heitzmann, 2023-07-27:
Hi @the-dave-power, unfortunately that's not so straightforward. You could try breaking your boolean operation in smaller sets if that's an option. For example, dividing your geometry in a few rectangular windows and operating on each window separately to reduce the total number of vertices used in each operation.

the-dave-power, 2023-07-31:
Thanks @heitzmann 

I understand what you are saying and I'll give it a try