heitzmann/gdstk #194: polygon can not identify outliner and inliner

rrjia, 2023-07-27:
```
cutout = gdstk.Polygon(
    [(0, 0), (5, 0), (5, 5), (0, 5), (0, 0), (2, 2), (2, 3), (3, 3), (3, 2), (2, 2)]
)
```
from this code , we can read and write gds file, but we can not kown which is outer boundary which is inner  boundary, but in pyhthon shapely library it use this mothod:

```
shapely.Polygon
class Polygon(shell=None, holes=None)
A geometry type representing an area that is enclosed by a linear ring.
A polygon is a two-dimensional feature and has a non-zero area. It may have one or more negative-space “holes” which are also bounded by linear rings. If any rings cross each other, the feature is invalid and operations on it may fail.
```

did our library can do this? thanks

heitzmann, 2023-07-27:
No, polygons in gdstk are defined by an external boundary only. Support for holes is limited to what's shown in the [getting started](https://heitzmann.github.io/gdstk/gettingstarted.html#holes) section of the docs.

closed 2023-07-27 by heitzmann

rrjia, 2023-08-15:
```
cutout = gdstk.Polygon(
    [(0, 0), (5, 0), (5, 5), (0, 5), (0, 0), (2, 2), (2, 3), (3, 3), (3, 2), (2, 2)]
)
```
if we have holes, the external boundary will have duplicate point as show in above (0, 0), but when we have 2 holes, what should i do ? thankds

heitzmann, 2023-08-19:
@rrjia You just have to keep linking the holes to the external boundary. If you want to have it done automatically (or want to see one example to understand the process), you can use a boolean operation:

```python
In [1]: import gdstk

In [2]: polygons = gdstk.boolean(gdstk.rectangle((0, 0), (5, 5)), gdstk.rectangle((2, 2), (3, 3)), "not")

In [3]: polygons[0].points
Out[3]:
array([[5., 5.],
       [0., 5.],
       [0., 2.],
       [2., 2.],
       [2., 3.],
       [3., 3.],
       [3., 2.],
       [2., 2.],
       [0., 2.],
       [0., 0.],
       [5., 0.]])
```

For 2 holes:
```python
In [1]: import gdstk

In [2]: polygons = gdstk.boolean(
   ...:     gdstk.rectangle((0, 0), (10, 5)),
   ...:     [gdstk.rectangle((2, 2), (3, 3)), gdstk.rectangle((7, 7), (9, 9))],
   ...:     "not",
   ...: )

In [3]: polygons[0].points
Out[3]:
array([[10.,  5.],
       [ 0.,  5.],
       [ 0.,  2.],
       [ 2.,  2.],
       [ 2.,  3.],
       [ 3.,  3.],
       [ 3.,  2.],
       [ 2.,  2.],
       [ 0.,  2.],
       [ 0.,  0.],
       [10.,  0.]])
```