heitzmann/gdstk #197: invalid record type on global level

Labjoker, 2023-08-07:
hello,
When I use the library.write_oas() function to write a layout with 66667*500 graphics into an OAS file, there are no errors. However, when I write a layout with 66667*66667 graphic units and open the OAS file using Klayout software, it shows an error message "invalid record type on global level(position=8539813,cell=)". Moreover, the file size is only 8331KB, even smaller than the first time. May I ask what could be the possible cause of this error?

Thanks!

heitzmann, 2023-08-19:
Hi @Labjoker, sorry for taking long to respond, I've been quite busy lately and don't have as much time to focus on gdstk right now.

I have to run tests over here to see what I can find. If you can provide an easy way to reproduce the problem, it would be of great help.

Labjoker, 2023-08-21:
Hi, @heitzmann, thank you for taking the time to reply to me. 
The file I have written is a layout with 66667*66667 units, where each unit is a rectangular shape of size 400nm*100nm (with different rotation angles), and there is a periodicity of 600nm between units. You can try writing such a large file and then test its size and functionality once it's completed. If needed, I can share my program with you. 
Thank you again.

Labjoker, 2023-08-21:
> Hi @Labjoker, sorry for taking long to respond, I've been quite busy lately and don't have as much time to focus on gdstk right now.
> 
> I have to run tests over here to see what I can find. If you can provide an easy way to reproduce the problem, it would be of great help.

> Hi @Labjoker, sorry for taking long to respond, I've been quite busy lately and don't have as much time to focus on gdstk right now.
> 
> I have to run tests over here to see what I can find. If you can provide an easy way to reproduce the problem, it would be of great help.

Hi, @heitzmann , thank you for taking the time to reply to me.
The file I have written is a layout with 6666766667 units, where each unit is a rectangular shape of size 400nm100nm (with different rotation angles), and there is a periodicity of 600nm between units. You can try writing such a large file and then test its size and functionality once it's completed. If needed, I can share my program with you.
Thank you again.

Labjoker, 2023-08-21:
> Hi @Labjoker, sorry for taking long to respond, I've been quite busy lately and don't have as much time to focus on gdstk right now.
> 
> I have to run tests over here to see what I can find. If you can provide an easy way to reproduce the problem, it would be of great help.

Hi, @heitzmann, thank you for taking the time to reply to me.
The file I have written is a layout with 6666766667 units, where each unit is a rectangular shape of size 400nm100nm (with different rotation angles), and there is a periodicity of 600nm between units. You can try writing such a large file and then test its size and functionality once it's completed. If needed, I can share my program with you.
Thank you again.

heitzmann, 2023-10-07:
Sorry for the long delay @Labjoker, but I've been quite busy lately. I've tried to reproduce your error with the following test, but it seems to work fine for me:
```python
import gdstk

lib = gdstk.Library()
cell = lib.new_cell("MAIN")

n = 6667
for i in range(n):
    for j in range(n):
        poly = gdstk.rectangle((0, 0), (0.4, 0.1))
        poly.rotate(0.1 * (i ** (3 / 7) + j ** (2 / 3)))
        poly.translate(i * 0.6, j * 0.6)
        cell.add(poly)

lib.write_oas("large.oas")
```
It takes a while to run (97 s) and the resulting file is large (449 MB), as expected, but I opened it in KLayout without issues.

Could you share your code?