heitzmann/gdstk #208: [0.9.43 regression] Bundled qhull conflicts with pre-installed qhull

yurivict, 2023-10-09:
Files under ```/usr/local/include/libqhull_r``` like /usr/local/include/libqhull_r/geom_r.h conflict.

Is there a reason that you don'tm use separately installed qhull?

In case you really need to bundle the library - you need to move the bundled headers to a location unique to gdstk, like ```/usr/local/include/gdktk/libqhull_r/geom_r.h```.

Or, better, don't bundle and use the external qhull.

Version: 0.9.43

heitzmann, 2023-10-10:
I think the conflict is because those files should not be installed, so this looks more like a bug in our CMakeLists files than a problem with using a static version of qhull. I have to look into that.

heitzmann, 2023-10-12:
Ok. I've followed the same path as zlib for v.0.9.5. It's a little more troublesome because now we have to handle qhull installation in our CI wheel builders, but they are working now too.