heitzmann/gdstk #209: Reference cells are not written into the GDSII file. gdstk v0.9.46

mgolser, 2023-10-18:
If I add a reference cell to the top cell of the library, it is not written to the GDSII file. However, geometry that is added directly to the top cell does.

heitzmann, 2023-10-19:
Please, share an example of the issue so that we can understand the problem.

mgolser, 2023-10-19:
When I try the example from the documentation for the references, they are not written to the GDSII file. It doesn't matter whether I add it to a library as shown in the code and then write it or use the GdsWriter directly. If I save the cell with the references as an svg and then look at them, they are added. 
Code:
```python
import gdstk

ld = {
    "contacts": {"layer": 0, "datatype": 0},
    "cu": {"layer": 1, "datatype": 0},
    "gnd": {"layer": 2, "datatype": 0},
}
p1 = gdstk.rectangle((-3, -3), (3, 3), **ld["contacts"])
p2 = gdstk.rectangle((-5, -3), (-3, 3), **ld["contacts"])
p3 = gdstk.rectangle((5, -3), (3, 3), **ld["contacts"])
p4 = gdstk.regular_polygon((0, 0), 2, 6, **ld["contacts"])

contact = gdstk.Cell("CONTACT")
contact.add(p1, p2, p3, p4)

device = gdstk.Cell("DEVICE")

ref1 = gdstk.Reference(contact, (3.5, 1), magnification=0.25)
ref2 = gdstk.Reference(contact, (1, 3.5), magnification=0.25)
device.add(ref1, ref2)

main = gdstk.Cell("MAIN")
main.add(gdstk.Reference(device, (0, 0), columns=3, rows=2, spacing=(6, 7)))

lib = gdstk.Library()
lib.add(main)
lib.write_gds("test.gds")
```

[image]

heitzmann, 2023-10-19:
`Library.add` does not add dependencies automatically (please see the note in [the documentation](https://heitzmann.github.io/gdstk/library/gdstk.Library.html#gdstk.Library.add)). You have to also do `lib.add(*main.dependencies())`

mgolser, 2023-10-19:
That solves my problem, thank you very much.

closed 2023-10-19 by mgolser

github-young, 2024-03-25:
> `Library.add` does not add dependencies automatically (please see the note in [the documentation](https://heitzmann.github.io/gdstk/library/gdstk.Library.html#gdstk.Library.add)). You have to also do `lib.add(*main.dependencies())`

I think it would be better to explain this point explicitly. I encountered the same problem that I cannot see the referenced object array correctly in KLayout, and I have to search in the issues to solve it by adding `*main.dependencies(True)` mentioned in #237. However, this is not mentioned either in the [Getting Started](https://heitzmann.github.io/gdstk/gettingstarted.html#references) or in [the documentation](https://heitzmann.github.io/gdstk/library/gdstk.Library.html#gdstk.Library.add) you gave.