heitzmann/gdstk #214: Abutment of two cells

Rishabhgoyal07, 2023-11-23:
I am using gdstk.Reference() function to abut two cells side by side, and saving them into a cell, then adding the cell into a library, then finally saving the library as gds file, but it is not giving the abutted two cells as the output, infact it is showing nothing when viewed in klayout.

heitzmann, 2023-11-23:
Can you share your script? Note that gdstk does *not* add dependencies by default when adding a cell to a library, you have to manually add *all* your cells or do something like:
```python
library.add(main_cell, *main_cell.dependencies())
```

Rishabhgoyal07, 2023-11-24:
import gdstk

#  defining a function to abut two cells
def abutment(file1,file2):
    
#  loading the gds file into a library
    lib1 = gdstk.read_gds(file1)
    lib2 = gdstk.read_gds(file2)

#  to know the information about the gds file
    info1 = gdstk.gds_info(file1)
    print(info1)

#  extracting the cells from the library
    inst1 = lib1.cells[0]
    inst2 = lib2.cells[0]

#  creating new library to hold the abutted cell
    lib3 = gdstk.Library()
    # lib3.add(inst1)
    # lib3.add(inst2)

#  creating a reference of cell(instance)1  
    ref = gdstk.Reference(inst1)

#  creating a cell two hold the abutted cells
    inst3 = lib3.new_cell("inst3")

#  adding the referecne of cell1 in cell3 
    inst3.add(ref)

#  Writing the library to a gds file
    lib3.write_gds("inst1.gds")

#  Calling the function to abut the cells
abutment("inverter1.gds","inverter.gds")

This is my script, it is executing successfully but in the output it is showing nothing in the klayout, while I want to see the two cells which I referenced and added to inst3.

closed 2023-11-27 by Rishabhgoyal07