heitzmann/gdstk #216: Connecting two Cells together

Rishabhgoyal07, 2023-11-27:
I have layout of two cells, I have placed them side by side, but what can we connect them together like in cell1 there is Metal M3 running in top cell, how can this M3 can be connect to the M3 of cell2, so that both the cells are connected to each other when we place them together.

joamatab, 2023-11-27:
You can use [gdsfactory](https://gdsfactory.github.io/gdsfactory/notebooks/01_references.html#connect-references)

if you add ports is easy to connect cells together

Rishabhgoyal07, 2023-11-28:
does gdsfactory has functions to basically abut two cells and connect two cells according to the pins given in the layout?

@joamatab

tvt173, 2023-11-28:
@Rishabhgoyal07 there are also various routing functions in gdsfactory, if that's what you're looking for.

https://gdsfactory.github.io/gdsfactory/notebooks/04_routing.html

Rishabhgoyal07, 2023-11-28:
are these ports CAD layers?? used for only identification??

joamatab, 2023-11-28:
The ports are not drawn in CAD layers, but saved as python object properties

You can always add pins to the ports if you want to draw them in CAD layers

Rishabhgoyal07, 2023-11-30:
@joamatab  @heitzmann can i get the coordinates of a particular layer, if I give the layer number and datatype in gdstk??
Please reply to this question

nmz787-intel, 2023-11-30:
sounds like you want to create a new cell, let's call `cell1_2`, and you then reference `cell1` and `cell2` with their respective placement offsets. Then you can place `cell1_2` and it will give you the two cells next to each other. When you CAD makes it to the final die database, as long as there are no gaps between your example `M3`, most DRC/verification tools will merge shapes that touch, regardless of their cell hierarchy. As well, when you get to lithography, if there is no gap, they will end up as a single shape in resist/metal.

https://heitzmann.github.io/gdstk/library/gdstk.Reference.html

```
cell1= gdstk.Cell("cell1")
# add polygons
cell2= gdstk.Cell("cell2")
# add polygons
cell1_2= gdstk.Cell("cell1_2")
ref1 = gdstk.Reference(cell1)
ref2 = gdstk.Reference(cell2, origin=(10,0)) # ref2 is 10 units away in X dimension from origin
cell1_2.add(ref1, ref2)
# add all 3 cells to your library
lib = gdstk.Library()
lib.add(cell1, cell2, cell1_2)
```