heitzmann/gdstk #222: does not support gds.gz format

Larson-Li, 2023-12-13:
In actual project, the GDS file is large, so compress the gds to gds.gz is common actions, but the gdstk.read_gds does not support the action.

Larson-Li, 2023-12-21:
Does gdstk will support gz compress format when using gdstk.read_gds API?

heitzmann, 2023-12-21:
Not yet. It's a planned feature but there's no one working on it at this time.

Larson-Li, 2024-05-24:
@heitzmann Is there schedule of this feature?

heitzmann, 2024-05-24:
Not really. This feature is not trivial and I don't really know when I'll have the time to dedicate to it. I'm always open to PR's, though! Just make sure it is implemented in a C-style (not modern C++) to keep in line with the rest of the code base.

Larson-Li, 2024-12-20:
> Look at my pull request #286 where I have added this feature But I need help testing it

Gds and gz are golden formats, sample test is enough for merging. I will test and feedback after is approved