heitzmann/gdstk #223: Polygonal coordinate problem

SouthChinaHuihuisauce, 2023-12-14:
Hello dear author, I encountered a problem when using gdstk to draw polygons. When obtaining coordinates, it seems that gdstk will also output the defined polygon coordinates. I would like to know how to close it?

[image]
[image]

heitzmann, 2023-12-14:
You seem to bee looping over all cells in the library. You probably want to select 1 cell and only display polygons from that cell. In the KLayout screenshot you are only displaying polygons from the NMOS cell (the `get_polygons` function will get polygons from references, so there's no need to loop over cells.)

SouthChinaHuihuisauce, 2023-12-18:
> You seem to bee looping over all cells in the library. You probably want to select 1 cell and only display polygons from that cell. In the KLayout screenshot you are only displaying polygons from [...]

Thank you for your reply. I have resolved the issue through your reply. Thank you

SouthChinaHuihuisauce, 2023-12-18:
@heitzmann Hello, I would like to ask if there is any GDSTK that can output cell information, similar to Klayout.
[image]

SouthChinaHuihuisauce, 2023-12-18:
@heitzmann Dear author, hello. I have output all cell names, but I don't know how to distinguish the relationship between calls
[image]

heitzmann, 2023-12-21:
@SouthChinaHuihuisauce you have to look into the `references` array for each cell. The cells form a DAG structure that can be traversed if you go through each of the references.

SouthChinaHuihuisauce, 2023-12-25:
@heitzmann Dear author, first of all, thank you for your reply. I have tried the references you mentioned, but I am not familiar with gdstk and gds files. Can you give me an example?

[image]
[image]

heitzmann, 2023-12-28:
You'll probably want to write this as a recursive function. Something like this (in python for brevity):
```
def print_references(cell, indent=""):
    print(indent + "- " + cell.name)
    for reference in cell.references:
        print_references(reference, "    " + indent)
```

SouthChinaHuihuisauce, 2024-01-11:
@heitzmann  Hello dear author, due to a busy period of time, I apologize for not replying to your message. Recently, I tried the example you provided in Qt and found that there were no references when using cells in C++. Instead, I used the references in cells_ It seems that the array is not quite right. Could you please give me some guidance? Thank you very much.

`void MainWindow::print_references(gdstk::Cell &cell, QString indent)
{
    qDebug() << indent << "-" << cell.name;

    for (size_t i = 0; i < cell.reference_array.count; i++) {
        print_references(*cell.reference_array[i], "    " + indent);
    }
}`
[image]

SouthChinaHuihuisauce, 2024-01-11:
[image]

SouthChinaHuihuisauce, 2024-01-11:
@heitzmann Thank you very much for your sample code. We have found it

closed 2024-01-14 by heitzmann