heitzmann/gdstk #224: Memory allocation using regular repetitions.

arondahl, 2023-12-14:
Hi,

I noticed there is a large disparity of memory use, when generating reference repetitions using perpendicular v1&v2 vectors in gdstk.Repetition() versus using vectors, which are not perpendicular. I was wondering why this is and also whether there is a work around?

Best,
Aron Dahlberg

heitzmann, 2023-12-15:
Are you talking about runtime memory usage or the size of the resulting GDSII file? Both are expected to be larger in the non-orthogonal case, but the largest difference will be in the saved file, because the GDSII format does not support this type of array.

heitzmann, 2023-12-21:
Yes, in that case you have to use the OASIS format to see any gain.

closed 2023-12-21 by heitzmann