heitzmann/gdstk #228: Create binary distribution for Windows and Python 3.12

isaacansys, 2024-01-03:
The PyPI Packages workflow currently creates binary distributions for Linux and MacOS but not Windows, per https://pypi.org/project/gdstk/#files

Please update to include Windows/3.12 binaries, thank you!

daquinteroflex, 2024-04-17:
Hi @heitzmann , hope you're well! Unfortunately I think we need this update too per this action https://github.com/flexcompute/tidy3d/actions/runs/8718852385/job/23916936453?pr=1622 related to https://github.com/flexcompute/tidy3d/issues/1317

If I get a moment I'll see if I can help more on a PR, although you have more expertise on this!

heitzmann, 2024-04-17:
Done in 0.9.51!

closed 2024-04-17 by heitzmann

daquinteroflex, 2024-04-17:
Thanks a lot Lucas!

daquinteroflex, 2024-04-17:
Hi Lucas, sorry to bother you, still getting the same error:

```
  -- Found ZLIB: /Applications/Xcode_14.2.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/lib/libz.tbd (found suitable version "1.2.11", minimum required is "1.2.7")
  CMake Error at /usr/local/Cellar/cmake/3.29.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
    Could NOT find Qhull (missing: QHULL_LIBRARY QHULL_INCLUDE_DIR) (Required
    is at least version "8")
  Call Stack (most recent call first):
    /usr/local/Cellar/cmake/3.29.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
    cmake_modules/FindQhull.cmake:170 (find_package_handle_standard_args)
    src/CMakeLists.txt:23 (find_package)
  
  
  -- Configuring incomplete, errors occurred!
  
  *** CMake configuration failed
  

  at /Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/poetry/installation/chef.py:164 in _prepare
      160│ 
      161│                 error = ChefBuildError("\n\n".join(message_parts))
      162│ 
      163│             if error is not None:
    → 164│                 raise error from None
      165│ 
      166│             return path
      167│ 
      168│     def _prepare_sdist(self, archive: Path, destination: Path | None = None) -> Path:

Note: This error originates from the build backend, and is likely not a problem with poetry but with gdstk (0.9.51) not supporting PEP 517 builds. You can verify this by running 'pip wheel --no-cache-dir --use-pep517 "gdstk (==0.9.51)"'.
```

https://github.com/flexcompute/tidy3d/actions/runs/8723685631/job/23932467320?pr=1622