heitzmann/gdstk #229: Pin Identification and Extraction

Rishabhgoyal07, 2024-01-04:
@heitzmann , How can we identify the pins that are present in the leaf cells (bottom hierarchy) of a layout, and export them to the top level of the layout?? Is this possible in gdstk?

heitzmann, 2024-01-06:
How are the pins represented? There's no "pin" object in GDSII files. If they are paths or polygons, you can use `get_polygons` with the appropriate layer/datatype info. If they are nodes, gdstk doesn't have support for those, unfortunately.

Rishabhgoyal07, 2024-01-09:
I have created a **label** using the ### Label class, but when i am using the **cell.labels function** to get all the labels of a cell, the **created label** is not coming in the list, also when i am viewing my cell in cadence virtuoso, there also it is not getting recognized, why is this happening??

Rishabhgoyal07, 2024-01-12:
Thank you for your help, also i want to know how can we **flip a cell horizontally,** i..e we the **left side of the cell** becomes the **right side and the right side becomes the left side.**

heitzmann, 2024-01-14:
You can use a reference setting `rotation=180` and `x_reflection=True`.

Rishabhgoyal07, 2024-01-15:
I did that, but the quadrant in which the cells is, got changed from **1 to 4 quadrant**, i..e the x-axis is changed from **x to-x**. Is there any way so that the x-axis doesn't get changed, i..e, the **cell is mirrored in its place only**?

closed 2024-01-25 by Rishabhgoyal07

nmz787-intel, 2024-03-07:
> I am viewing my cell in virtuoso, in virtuoso we can directly create a pin on any metal, while the created pin using gdstk is not getting recognized in virtuoso.

You need to use the appropriate options during `strmin` or `oasisin`, such as `-objectMap` or `-pinAttNum`, or write `SKILL` or `OaScript` code to find your `Labels` on your `metal:pin` layer:type and then convert them into Pin objects